gerritv/Grbl-Panel #21: grbl-panel does not connect to ramps1.4

skeitzer1, 2015-02-04:
I found I needed to add a sleep(1000) between the set dtr and clear dtr which is used to reset the arduino.

This cleared my hang on connect. The arduino needs time to completely reset before any commands are issued to it..

Nice program..  using it to control my Shapeoko 2

Thanks

gerritv, 2015-02-04:
Interesting, is that over and above the Start up Delay that is configurable on the Settings page?

Someone else had reported a problem with Mega2560, which is why I added that setting.

electrokean, 2015-02-07:
I was having the same issue this weekend using Grbl-Panel with a Mega2560 (running a custom 6-axis version of Grbl I'm testing out).
If I monitor the serial data, I can see the "?" status enquiries start immediately, rather than waiting based on the start up delay.
The fix was be pretty simple:  In GrblGui.vb subroutine btnConnDisconnect_Click, move the line "statusPrompt("Start")" one line down so it is after the Sleep().
After I did this, I had no problems connecting.

electrokean, 2015-02-07:
It is a trivial change, but I've created a pull request

gerritv, 2015-11-03:
Finally integrating this into a release. Should be done before Nov 10

gerritv, 2015-11-20:
Fixed in Release 1.0.5.0
Thank you for your patience

closed 2015-11-20 by gerritv