gerritv/Grbl-Panel #23: Some new features..  inch position display and g code single step

skeitzer1, 2015-02-08:
Hi,

I added a checkbox to set the 3 axis position displays to inches or mm. Code change just divides the positions by 25.4 before setting the numerical text boxes for position display..  I'm an old 'inches' guy..  I thought there might be a user setting to set this, but I could not find it..  My grbl is working in inches but the display was always mm..

I also added another command button to the open/ run/ pause/ stop group box. The new button is "single step".  I was trying out a new g-code generator, and needed to see where things were going bad..  Single stepping thru the g code helped this process.  Again, not a rocket science addition, but helped me out.

I'm new to this 'community software' effort..  willing to work on problems/ feature ads/ share my code if anyone is interested..  just do not know how or what are 'the accepted methods/ process for doing so'.. Love the mixture of software/ making things work..

Thanks again for making a nice program..

gerritv, 2015-02-09:
Hi
thank you for the thanks and the suggestions.
you can get reports in Imperial by setting $13 to 1.
Can you email me your code for single step? Probably easier than figuring out Git :-)
3d58b8e6@opayq.com is my masked email address.

Gerrit

gerritv, 2016-12-15:
Finally I am getting to adding your Single Step code. Very cleanly done! Thank you

closed 2016-12-15 by gerritv