gerritv/Grbl-Panel #28: Position Display is blank

andre1234567891, 2015-08-12:
I am working on a small 2 axis project and of all the programs I have tried I like Grbl-Panel the best.
Just having 1 problem, the position display is blank.
Any idea on what could be the problem?

Andre

gerritv, 2015-08-13:
I have only tried Grbl-Panel with 3 axis so there might be something amiss.

Is your $10 param set to 15? 
Which version of Grbl are you running?
What does the Status window show?

andre1234567891, 2015-08-13:
You have my respect, I was not expecting a response so fast.

grbl_v0_9j_atmega328p_16mhz_115200
Downloaded from here a few days ago   http://bit.ly/1I8Ey4S
Programmed to the UNO with XLoader also downloaded a few days ago.

$10 is set to 18
I can change it to 15 but I am not finding a way to send it to grbl.
So when I exit Grbl Panel and restart $10 is back to 18

Grbl Panel version 1.0.3.0 build date 2015-08-11 08:18PM

Status is green and says IDLE
I can check the vebose box and get line after line saying
<idle,WPos:0.000,0000,0.000,Lim:000>

I can send a g code program and it runs fine.

^^^^^^^^^^^^^^^^^^^^^^^
Andre' B.

gerritv, 2015-08-13:
I'm retired so have lots of time :-) Not really, busier now than when I was working 12 hour days. :-) Seriously, I take support issues seriously.

Just to confirm, your Position display is empty/blank (not showing 0.000 on any axis)?

Setting a $ option changes requires you to double click the box with the new value in it. There is a Refresh button near the bottom of that section so you can verify the change.
My code doesn't yet handle the Lim although its presence shouldn't matter.

I will load the newest Grbl into my test device on Thursday to see if there is an issue.

andre1234567891, 2015-08-13:
Got it
About 2 minutes after I sent my last message the light bulb came on.
I typed $10=15 into the MDI box and hit return, the position display now 
shows the position and it matches the X,Y numbers in the g code.

Yes the display was empty.

Thank you very much.

Now on to the next problem.
Some background FYI my boss and I are quoting a small plastic part that 
has a bunch of small holes drilled at every possible angle.  We need 360 
degrees on the C axis and almost 300 degrees on the A axis to do the 
entire part in one setup.  I have never seen any commercial indexer or 5 
axis mill that would do it, no money for that anyway.
The required accuracy is low so I told the boss I would look into 
bolting some steppers together and see if I could sneak a long skinny 
tool past all the indexer parts to do the job.
The idea was to put a cycle start switch on the mill table and have the 
mill drill a hole then go hit the switch, the indexer would move to the 
next position and wait (M0) for the next cycle start.  The M30 at the 
end of the program would rewind and wait for a cycle start input.
When I seen the A2 pin labeled Cycle Start/Resume I figured all was good.
https://github.com/grbl/grbl/wiki/Connecting-Grbl

Now the more reading I do the less likely it seems that the cycle start 
input will actually start the cycle.  And every g code sender I look at 
has to reload the program to send it again instead of rewinding at the M30
So may have to move on to plan B, when working with my boss you need 
Backup plans :)
The backup would be to take the UNO back to its native form and write a 
sketch using one of the stepper motor libraries.

I will need the grbl stuff for a personal project anyway, and I will 
very likely have more questions, just need to find some time between 
work projects.

Thanks,

^^^^^^^^^^^^^^^^^^^^^^^
Andre' B.

gerritv, 2015-08-14:
Backup plans are always a good thing to have.

I can try adding a rewind for M30. But that leaves open how you get the machine to tell GrblPanel that a Cycle Start is required?
I suppose this is urgent? What is the time frame?

andre1234567891, 2015-08-14:
Not real urgent.  We have not actually gotten the job yet but it does 
look like it will happen.  And we were told the first few orders would 
be small 5 to 10 parts, so the software as it is now would work.  Also 
since the indexer would be new and unproven I would be running the job, 
not just setting it up then handing it off to an operator like I 
normally would.

I have some ideas I would like to try and would like to get back to PC 
programming, it is just that the last program I wrote was for Windows 
3.0, since then it has been all CNC machines, everything I needed to do 
on a PC has been in VBA in Word and Excel.

Looks to me like if the first line of a gcode program was a comment with 
something like "(ESC-AUTOSTART)" and when Grbl Panel read that it sent a 
"M0" to Grbl that would put Grbl into the run state, waiting for a 
resume input.

As for the M30 command, as some of the CNC machines I program are over 
40,000 pounds and have servos in the 15 to 20 HP range some caution is 
in order.
But then for most anything controlled by Grbl, probably not a problem.
I just tried this.
Load up a program and start it running.
Load up the same file in a text editor, change it and save it.
Go back to Grbl Panel and finish the program.
Then when you reload it to run again, surprise it is not the same program.

At least with things as they are now the operator is responsible for it 
being the right file by having to manually reload the file.
I am thinking that especially on a network where other people may have 
access to the file while you are running  an auto reload could be a problem.

With the implementation of the M30 it may be better to make a temp copy 
of the file when the user opens it and then run and rerun from the copy.

^^^^^^^^^^^^^^^^^^^^^^^
Andre' B.

andre1234567891, 2015-08-19:
Well, I have managed to download an get Visual Studio Community 2013.  
Not sure if I needed everything it has.
And I was able to compile and run the code found at.
https://github.com/gerritv/Grbl-Panel/archive/master.zip

I also made a few changes and got an autorun hackjob going.
I am much more used to C programming and this object stuff gets me 
scratching my head at times.

^^^^^^^^^^^^^^^^^^^^^^^
Andre' B.

gerritv, 2015-08-21:
Mnn, I missed your Aug 14 message :-(
If you send me your changed code I can maybe sanitize and include for others? (Use the email address listed with my user name)
I had implemented a ReLoad UI button, not sure if that is what you needed. A 'proper' M30 implementation might be to just reuse what was already loaded (not reloading from the source file), thus avoiding/preventing the problem you raised about others tinkering with the file.

(Everything seems to compile under VS2015 Community Edition as well.)

gerritv, 2015-09-01:
I will be trying to merge your code over the next few days.

gerritv, 2015-11-03:
I am finally integrating the M30 support. Not sure yet if I will add your {ESC} handling yet. Would mapping a keyboard key to Run/Start work for you instead? This you can use with an external pad, e.g. Xbox One game pad?

gerritv, 2015-11-20:
M30 handling is in V1.0.6.0. When an M30 is detected, the gcode view is cleared of status and the Send button is re-enabled. The file is NOT reloaded, thus preventing unexpected behavior if someone edits the file during a run.

closed 2015-11-20 by gerritv