gerritv/Grbl-Panel #29: homing cycle, zero xyz problem

cabinetguy, 2015-09-24:
I'm relatively new to the CNC world and finished assembling my c-beam machine last week. I got it working fine by following the 2 hour tutorial and was able to successfully cutout a couple of items I designed in Sketchup.

The problems started when I installed my limit switches and tried to do a home cycle. The initial tutorial said that the software considers the front, left corner as the zero xyz point. When I enabled homing, it runs all motors in their positive direction which lands the router in the back right corner with the router up as high as it can go. When I tried to run those same files again, the alarm went off immediately as the machine hit its X limit immediately. Apparently the machine didn't get the memo that our new homing direction was the opposite corner. Any ideas on how to fix this?

langwadt, 2015-09-24:
afaik that is standard behavior for grbl, after homing the machine space is all negative coordinates

wlaw, 2015-11-14:
Before I did a large upgrade to my machine, grbl would home to the top right of the machine. This was very convenient because it moved the cutter out of the way to load material. After the rebuild, I told grbl to home at the bottom left corner. 

This is complete by changing $23=3 (homing dir invert mask:00000011).
So, for my machine 00000000 sent the home to the top right. +X, +Y
00000011 sends my machine to bottom left corner, -X, -Y.

Hope this helps

cabinetguy, 2015-11-17:
That helped!  Thanks for the reply

gerritv, 2015-11-20:
Ok to close?

closed 2015-11-20 by gerritv