gerritv/Grbl-Panel #30: Error connecting to XCarve (Arduino Uno, Gshield, Grbl 0.9j)

RealAbsurdity, 2015-10-28:
I am eager to try Grbl-Panel but cannot get the arduino for my xcarve to connect. Tried disabling the polling and cranking up the startup delay, as suggested in previous issues. After rescan the com port usually comes back with some wonky character following the port number. Tried both the scanned string and just typing COM3 to no avail. Any ideas? No trouble connecting with UGCS, but would like better control over the machine.
[image]

RealAbsurdity, 2015-10-28:
HA! Fixed it. Apparently a windows 10 issue. Ran it in Windows 7 Compatibility Mode and it is up and running. FYI.

gerritv, 2015-10-29:
I was going to ask which OS. I will wake up my machine on Thursday (after >9 months asleep!!). Glad you have a workaround, will figure out the real fix in code.

gerritv, 2015-10-30:
I tried tonight with Grbl 9h and it worked fine?? So tried to install 9j and can't load it up!
What did you use to load Grbl onto your device? Xloader no longer works for me. On Windows 10, was fine on Windows 8.1.

RealAbsurdity, 2015-10-30:
The device came pre-installed with grbl from instructables. They recommend
xloader on their help page.

kram242, 2015-10-30:
Hello Gerrit, 
On my xPRO board that I show on the video it comes pre-loaded with GRBL already so I did not go into doing the firmware update.
In the past I have used xloader to install the firmware and for the xpro you can find out more on how to do this here.
https://github.com/Spark-Concepts/xPRO/wiki/3.-Updating-GRBL-Firmware

hope this helps but if you still can't get it you could ask the xpro guys in their support forum here as they may know better.
http://www.spark-concepts.com/forums/

Thank you
OpenBuilds Team

From: Gerrit 
Sent: Thursday, October 29, 2015 8:45 PM
To: gerritv/Grbl-Panel 
Subject: Re: [Grbl-Panel] Error connecting to XCarve (Arduino Uno, Gshield, Grbl 0.9j) (#30)

I tried tonight with Grbl 9h and it worked fine?? So tried to install 9j and can't load it up!
What did you use to load Grbl onto your device? Xloader no longer works for me. On Windows 10, was fine on Windows 8.1.

gerritv, 2015-11-20:
I can't reproduce this problem so can't fix it. And thankfully you have a 'fix' to keep you running. If it appears again please re-open this issue
Thanks

closed 2015-11-20 by gerritv