gerritv/Grbl-Panel #31: When using the keyboard to move the cutter head, as soon as i click on something else the keyboard movement stops working.

wlaw, 2015-10-31:
First, I think this is the best sender that i have used. I has great possibilities but the keyboard stops working for controlling the machine. I need to use a pendant because the computer is to far away to move the check then move then check. I can use the program Xpadder to set keyboard calls to buttons on a USB gaming pad. This allows me to stand in front of my machine and set Zero's, change rates. But the keyboard calls seem to lose focus when I move into another part of the program. 

All of this can be done on UGS, but i like this program better.

gerritv, 2015-11-01:
I can't see anything obvious in the code that would prevent doing this. And I don't have a gamepad to test it with :-(
To help me narrow this down: If you use this with Set Work X to Zero, when does it work and where is the focus when it no longer works? What action are you setting in the Xpadder config for this mapping?

wlaw, 2015-11-01:
When I go into Settings - I enable the keyboard arrows to move.
At first the arrows will move the machine, then they stop as soon as I use the mouse, on the panel.

The block is still checked in settings.

I have to un-check use keyboard arrows in settings.
exit, re-enter and then check the block and it will work again- until I use the mouse on button. 

In UGS - I have set Increase step and decrease step (divide and multiply on right key pad) to left and right triggers. 

UP, DOWN, LEFT and Right to the arrow keys, and Z up and Z down on two buttons.

I can also assign a button to reset X,Y,Z zero

All I can see on the G-Panel is arrow keys.

Can I get them to keep working?
What other keys can I use? Z - UP/Down, Zero, Increase/Decrease step?

[image]

I know people have used Xpadder and Xbox controllers but all I had was a MS sidewinder game pad. 

[image]

As I said before, I know it will work, if I can get the keyboard keys to stay working?
Thanks and great work!

wlaw, 2015-11-01:
Also,

Link to older free version of Xpadder - 
 http://www.techspot.com/downloads/5869-xpadder.html

Im using free version in Windows 7 - Running compatabile with XP rev 3.

wlaw, 2015-11-01:
I just noticed that the arrow keys will move the selection in the jogging box. 
If i select a different value for Distance, the arrow keys will move through the distance values.
Same with feed rate and jogging, so the arrow keys get used to move the selected value.

If you could set Y+ to keypad 8,

gerritv, 2015-11-01:
Thank you for the clarifications. I will try to find an Xbox 360 controller locally.
In meantime I will see about assigning specific codes. I have been away from this for quite a while so please be patient while I get my brain back in gear :-)

gerritv, 2015-11-01:
Can you email me the xpadder config file you are using for UGS? 3d58b8e6@opayq.com 
thanks

wlaw, 2015-11-01:
Just to be clear, the Xpadder just assigns a keyboard key to a joystick, or game pad button. Any usb controller with the correct drivers should work. 

The one I sent to you, is for the Sidewinder USB game pad. it controls arrow keys, /, *, +, - on the keyboard keypad.

The X and Y work on the Grbl Panel also with no changes.

gerritv, 2015-11-01:
I can reproduce the situation, now just have to figure out why I stop getting the key down events in the code.

wlaw, 2015-11-02:
gerritv, 
I just also found out by looking at the code we could move the Z-Axis with page up and down. This will work all of the time. So, my suggestion is to use the keypad keys on the keyboard instead of the arrows. Use them like the PgUp, PgDn buttons or don't use the arrow keys in distance and federate.

gerritv, 2015-11-02:
It is unfortunately not that simple :-( .Net uses the arrow keys to move about in a control once that control gets focus, as you are seeing with the distance radio buttons. The usual method of looking at keyboard events is then bypassed, so I no longer see those arrow keydown events. Hence your issue once you click elsewhere than the jogging section.
Also, the keypad keys have the same key code as non-keypad keys. And even if they were different, not everyone has keypads.
I'm in process of coding up an IMessageFilter, this captures events before any control or form gets to see it and act. This will let me keep arrow keys as jogging only.
The second part of your request involves defining keyboard mapping for most of the GUI buttons that are likely of interest for use by a dongle/game pad.
I expect to get most of the coding done Monday morning. Once I have something tested I will email you a zip file with new exe to try out.

wlaw, 2015-11-17:
gerritv,

I was just wondering if you made any progress on this.
Not pushing, I still think its the best.

gerritv, 2015-11-17:
I was testing the changes yesterday, should be on GitHub on Tuesday. I added mappings for X, Y, Z and All reset to 0 as well.

wlaw, 2015-11-17:
The only other mapping I would request is a Key to increase or decrease steps size. It would be great if you could map specific keys to .01mm, .1mm, 1mm and 10mm. But a button that advances to the next higher level would be fine to.  This is only a request, please send what you have, it will make my life much better. 
The older I get the harder it is to see that little end mill way over there :-)
Thanks @gerritv.

gerritv, 2015-11-19:
Check your email, I sent you a version. Add, Subtract, Multiply and Divide operate the step size etc. I forgot that I already added those. Should be same as UGS

wlaw, 2015-11-19:
Very nice. I will test it this weekend, but I never received the file. Can
you try one more time?

Check your email, I sent you a version. Add, Subtract, Multiply and Divide
operate the step size etc. I forgot that I already added those. Should be
same as UGS

wlaw, 2015-11-20:
I am not getting this file, is there another place I can get my contact info to you? Or am i just missing something like github has the file somewhere?

gerritv, 2015-11-20:
I used the xpadder email and replied to it. I am working on refreshing my mind on using GitHub to publish the code changes today.

gerritv, 2015-11-20:
This is now in Release V1.0.5.0, see Releases above
Wiki will be updated shortly with keymapping info

Thank you for reporting the issue and the suggestions. And lastly thank you for your patience

closed 2015-11-20 by gerritv