gerritv/Grbl-Panel #33: Homing cycle button missing.

wlaw, 2015-11-21:
Testing out version 1.0.6 it seems the homing button is missing at least on mine.

I can still call $H from the command line.

Thanks for the Great work @gerritv

gerritv, 2015-11-21:
Mmn, it is there on my copy. It doesn't show up until you connect to Grbl. It will then be Red until you Unlock.

wlaw, 2015-11-22:
I went back to 1.0.3 and it showed up again. When I switched to 1.0.5 and 1.0.6 it did not show up. I get the unlock button, but no home.
[image]

gerritv, 2015-11-22:
Puzzling, what OS are you running on?

wlaw, 2015-11-22:
7 Home Premium.
If connect it don't show.
[image: Inline image 1]

If i go into settings, and get get grbl settings..
[image: Inline image 2]
Then return

[image: Inline image 3]

Its back.

So, it looks like something to do this the initial read of the grbl
settings.

once i get the settings, it shows up.

Connect on load don't help.

gerritv, 2015-11-22:
Can you post the output of the Help/About/Details/Assemblies list?

wlaw, 2015-11-22:
[image]

Here you are

gerritv, 2015-11-25:
What is your screen resolution?
I have built a Win7 machine, tested ok so far but the .Net files are all from 2010. Waiting to update then will verify again as the current .Net files are much newer.

gerritv, 2015-11-29:
finished updating, still can't reproduce this :-(
What is your screen resolution, perhaps that is related.

wlaw, 2015-12-05:
Sorry for the delay, got a bit busy!
1280 x 1024 is my screen resolution.
That hasn't changes since i have had this computer.

Thanks for helping

wlaw, 2016-01-03:
Well, I can get the home button to show up every now and again but nothing permanent. As I explained before it works fine on the 4th release,  release 5 no joy.

gerritv, 2016-01-27:
Can you try the current version? Code is now on .Net 4.5.2

wlaw, 2016-02-02:
Just a thought about this.

I did try the new version and Net 4.5.2 Still not loading.

What if you were to load grbl settings on open or connect..
"If i go into settings, and get get grbl settings.."

Is that easy to do or no.

gerritv, 2016-02-02:
Good plan, I will look into it.

gerritv, 2016-02-07:
I checked the code, I load ($$) the settings on Connect. Only if $22=1 then the Home button gets enabled.
Can you disable polling, enable Verbose and let me know if you see the $xx settings scroll by?

wlaw, 2016-02-07:
Started the program V 1.0.8.0 and connected.
[image: Inline image 1]
Go To Settings
[image: Inline image 2]
Press: Get Grbl Settings
[image: Inline image 3]
Back to main screen
[image: Inline image 4]

wlaw, 2016-02-07:
So disabled polling, enabled Verbose closed and restarted Version 9. I did not see the $$ but as soon as I entered the $$ in command prompt the homing button showed up.

This is not a problem for me as I know how to use the $H $X and all of the commands. Just more puzzling. I keep thinking its my computer, shield or UNO but when I load older versions it works great.

How do I transfer my Macros

gerritv, 2016-02-07:
I am not seeing your screen shots. Can you email them to me?
Re: moving the macros, I would open both versions of GrblPanel and copy/paste into the new one? Editing the .config file directly is problematic due to the embedded vbCr/vbLf in the macros.

wlaw, 2016-03-20:
OK, So after upgrade to Windows 10, my home button is back.

closed 2016-03-20 by wlaw

gerritv, 2016-03-26:
Glad that is finally resolved.