gerritv/Grbl-Panel #34: Ports scanning issue

neonflux75, 2015-11-27:
I've the Arduino connected on COM4 and I can't connect on the first time when program start cause of wrong port name.
In fact the dropdown menu doesn't show COM4 but COM4 with another strange random character like COM4漢 so I can't connect cause it says that the port doesn't exist.
I've to rescan several times before i can get only COM4 showing and at least each rescan the random character change. I've tried to launch the program with amministrator right too but unsuccessful.
It's my issue or others in my situation?

closed 2015-11-27 by neonflux75

gerritv, 2015-11-29:
Are you running Windows 10?
Please copy/paste the Help/About/Details/Assemblies, it might contain a clue for me.
The workaround for Windows 10 seems to be to enable Win 7 Compatibility mode. I am also running Windows 10 (64 bit) but so far unable to reproduce the problem.

neonflux75, 2015-11-30:
Yes I'm running Windows 10 (x64 too) and the fix that worked for me is to run the software in Windows 7 compatibility mode.
Sorry but I've not wrote down about that cause I've read this on one closed post in the Pulse tab. (https://github.com/gerritv/Grbl-Panel/issues/30)
Hope this helps :D