gerritv/Grbl-Panel #36: Grbl-Panel will not find any com ports win 10

joelaw419, 2015-12-05:
I have been using grbl-Panel on my win 8.1 machine for a long time and very pleased but I purchased a new win 10 computer and now it cannot find any com ports?

Grbl controller and Shapeoko software both work without problems but I don't want to use that software.

I want Grbl-Panel to work again.

Any suggestions.

Joe L.

gerritv, 2015-12-05:
Hi Joe
While I can't reproduce it myself, others have found that running Grbl-Panel in Windows 7 Compatibility mode solves this problem.
I wish I knew what the underlying issue is so I can fix it :-(

joelaw419, 2015-12-05:
Thanks gerritv for quick reply..

I am using a Shapeoko on a Win 10 Pro X64 bit Dell Laptop and Desktop.  I believe it is using FTDI but not sure.

I tried the compatible-mode and it seems to work OK on my Dell Desktop but not on my Laptop?.   I use Mach3 on my router and this software makes it more like what I am used to.  Very glad you did this but I still have a problem on the laptop.

How can I find type of USB or anything else that might work?

Joe L.

closed 2015-12-05 by joelaw419

gerritv, 2016-01-18:
V1.0.7.0 should mean that you no longer have to set Compatibility mode on Windows 10