gerritv/Grbl-Panel #37: Q: alarm when sending manual probe command

joebananas10, 2015-12-05:
I'm sending a manual probe command "G38.2 Z-50 F10" and sometimes the system returns a
"ALARM: Hard/Soft limit" 
but I am not on any limit. i can still move in both plus and minus Z. See attached image,
[image]
Sometimes this manual probe command works with no problem and other times it does not. I can't figure out what causes the difference. How can I find out which limit it's alarming on?

Please advise.

gerritv, 2015-12-09:
Hi Joe
Grbl 0.9j has some extra status bits to show the causes of Alarms. I am waiting for a new Uno to arrive, my old one didn't want to go into programming mode anymore :-(
Once I receive it I can add support for the new status' into Grbl Panel. In mean time set Verbose to see the stream coming back from Grbl.

gerritv, 2015-12-10:
$10 needs to be 31 in order to also report Limit pin status. I will figure out a place to display these, probably on RH side where the other Status items are displayed.

joebananas10, 2015-12-11:
Thanks for the info. I just checked my machine and $10 was set to 255. I changed it to 31 in the Settings grid.

closed 2015-12-11 by joebananas10