gerritv/Grbl-Panel #44: Open gcode very large file, wait a long time to prepare.

viewsat, 2016-02-03:
Open gcode very large file, wait a long time to prepare. About 100 million lines of gcode.

gerritv, 2016-02-03:
I have made a few changes in the code to reduce the time (will release soon) but I doubt it will help with that size of file :-(

I would have to re-write the file handling code to not buffer the whole file, not sure when I could do this.

viewsat, 2016-02-04:
Thank you for your quick response.
Grbl-Panel is a gcode sender in windows operating system is best used.

I opened this gcode test of 97 million lines to be tested.
http://www.picengrave.com/downloads/darklylabs/Roy%20Rogers%20Metric.zip

When I use Grbl-Panel Open this gcode test wait 3 minutes to read gcode, use 570MB of memory.

When I use the latest Universal GcodeSender open this gcode test, wait 10 seconds to read gcode, use 61MB memory.

gerritv, 2016-02-04:
Ok, I have your RoyRogers file loading in 10.3 seconds. A bunch of other things to adapt to the new file load method. Maybe do a release next week..

wlaw, 2016-02-05:
I just want to say, the response from @gerritv has been great. Your grbl panel is the best sender I have run across. While viewsat seems to be condescending in tone, I believe you are doing a great job in all the hassle that you go through for us. Thank You! 

Be respectful in what you say, this is being done as a gift!

wlaw

rwensley, 2016-02-05:
In response to @wlaw ....Here Here!

gerritv, 2016-02-05:
Thanks guys, for the thanks. I am enjoying this. More fun than when I was designing sw for a living.
Real soon now I will have my own machine running again, then I can enjoy this program as well :-)

I am not worried about the memory aspect, the loading speed though is an issue for the large files. I have made some excellent progress on that, more to come next week.

gerritv, 2016-02-07:
Fixed in V1.0.9.0, time to load the RoyRogers file is now <5 seconds.

closed 2016-02-07 by gerritv