gerritv/Grbl-Panel #45: Grbl Setting empty

pflugj, 2016-02-03:
The Grbl Settings are empty and when trying to "Get Grbl Settings" i get the following error message.

Unhandled exception has occurred in your application. If you click Continiue ....
Column "ID' is constrained to be unique. Value '$0'is already present.  ---- remark :The '$#'changes to different values.

The details Window reveals:  
===============================================================================See the end of this message for details on invoking 
just-in-time (JIT) debugging instead of this dialog box.

************** Exception Text **************
System.Data.ConstraintException: Column 'ID' is constrained to be unique.  Value '$0' is already present.
   at System.Data.UniqueConstraint.CheckConstraint(DataRow row, DataRowAction action)
   at System.Data.DataTable.RaiseRowChanging(DataRowChangeEventArgs args, DataRow eRow, DataRowAction eAction, Boolean fireEvent)
   at System.Data.DataTable.SetNewRecordWorker(DataRow row, Int32 proposedRecord, DataRowAction action, Boolean isInMerge, Int32 position, Boolean fireEvent, Exception& deferredException)
   at System.Data.DataTable.InsertRow(DataRow row, Int64 proposedID, Int32 pos, Boolean fireEvent)
   at System.Data.DataRowCollection.Add(Object[] values)
   at GrblPanel.GrblGui.GrblSettings.FillSettings(String data) in C:\Users\gerrit\Documents\My Projects\GrblPanel-0.1.0.6\Grbl-Panel\GrblSettings.vb:line 69
   at GrblPanel.GrblGui.showGrblStatus(String data) in C:\Users\gerrit\Documents\My Projects\GrblPanel-0.1.0.6\Grbl-Panel\GrblStatus.vb:line 206

************** Loaded Assemblies **************
mscorlib
    Assembly Version: 2.0.0.0
    Win32 Version: 2.0.50727.8670 (QFE.050727-8600)
    CodeBase: file:///C:/Windows/Microsoft.NET/Framework64/v2.0.50727/mscorlib.dll
----------------------------------------
GrblPanel
    Assembly Version: 1.0.3.0
    Win32 Version: 1.0.3.0
    CodeBase: file:///C:/Users/locke/Documents/C-Beam%20Machine/CBeam%20Machine%20Control%20Software/GRBL%20Panel/GrblPanel-Release-1.0.3.0/GrblPanel.exe
----------------------------------------
Microsoft.VisualBasic
    Assembly Version: 8.0.0.0
    Win32 Version: 8.0.50727.8670 (QFE.050727-8600)
    CodeBase: file:///C:/WINDOWS/assembly/GAC_MSIL/Microsoft.VisualBasic/8.0.0.0__b03f5f7f11d50a3a/Microsoft.VisualBasic.dll
----------------------------------------
System
    Assembly Version: 2.0.0.0
    Win32 Version: 2.0.50727.8670 (QFE.050727-8600)
    CodeBase: file:///C:/WINDOWS/assembly/GAC_MSIL/System/2.0.0.0__b77a5c561934e089/System.dll
----------------------------------------
System.Windows.Forms
    Assembly Version: 2.0.0.0
    Win32 Version: 2.0.50727.8670 (QFE.050727-8600)
    CodeBase: file:///C:/WINDOWS/assembly/GAC_MSIL/System.Windows.Forms/2.0.0.0__b77a5c561934e089/System.Windows.Forms.dll
----------------------------------------
System.Drawing
    Assembly Version: 2.0.0.0
    Win32 Version: 2.0.50727.8670 (QFE.050727-8600)
    CodeBase: file:///C:/WINDOWS/assembly/GAC_MSIL/System.Drawing/2.0.0.0__b03f5f7f11d50a3a/System.Drawing.dll
----------------------------------------
System.Runtime.Remoting
    Assembly Version: 2.0.0.0
    Win32 Version: 2.0.50727.8670 (QFE.050727-8600)
    CodeBase: file:///C:/WINDOWS/assembly/GAC_MSIL/System.Runtime.Remoting/2.0.0.0__b77a5c561934e089/System.Runtime.Remoting.dll
----------------------------------------
System.Configuration
    Assembly Version: 2.0.0.0
    Win32 Version: 2.0.50727.8670 (QFE.050727-8600)
    CodeBase: file:///C:/WINDOWS/assembly/GAC_MSIL/System.Configuration/2.0.0.0__b03f5f7f11d50a3a/System.Configuration.dll
----------------------------------------
System.Xml
    Assembly Version: 2.0.0.0
    Win32 Version: 2.0.50727.8670 (QFE.050727-8600)
    CodeBase: file:///C:/WINDOWS/assembly/GAC_MSIL/System.Xml/2.0.0.0__b77a5c561934e089/System.Xml.dll
----------------------------------------
RepeatButton
    Assembly Version: 1.0.0.0
    Win32 Version: 1.0.0.0
    CodeBase: file:///C:/Users/locke/Documents/C-Beam%20Machine/CBeam%20Machine%20Control%20Software/GRBL%20Panel/GrblPanel-Release-1.0.3.0/RepeatButton.DLL
----------------------------------------
Accessibility
    Assembly Version: 2.0.0.0
    Win32 Version: 2.0.50727.8670 (QFE.050727-8600)
    CodeBase: file:///C:/WINDOWS/assembly/GAC_MSIL/Accessibility/2.0.0.0__b03f5f7f11d50a3a/Accessibility.dll
----------------------------------------
System.Core
    Assembly Version: 3.5.0.0
    Win32 Version: 3.5.30729.8693 built by: QFE
    CodeBase: file:///C:/WINDOWS/assembly/GAC_MSIL/System.Core/3.5.0.0__b77a5c561934e089/System.Core.dll
----------------------------------------
System.Data
    Assembly Version: 2.0.0.0
    Win32 Version: 2.0.50727.8677 (QFE.050727-8600)
    CodeBase: file:///C:/WINDOWS/assembly/GAC_64/System.Data/2.0.0.0__b77a5c561934e089/System.Data.dll
----------------------------------------

************** JIT Debugging **************
To enable just-in-time (JIT) debugging, the .config file for this
application or computer (machine.config) must have the
jitDebugging value set in the system.windows.forms section.
The application must also be compiled with debugging
enabled.

For example:

<configuration>
    <system.windows.forms jitDebugging="true" />
</configuration>

When JIT debugging is enabled, any unhandled exception
will be sent to the JIT debugger registered on the computer
rather than be handled by this dialog box.
==============================================================================

I had value showing up when I started the program the first time.
Then I clicked something and then the settings disappeared and I got this error.

I have rebooted/-run ccleaner/ cleared cashe and deleted and copied the files back into place.
...so far no luck

Any Ideas what I should do to get it running?

Thanks for the nice program and helping.
Best regards
Jurgen

gerritv, 2016-02-03:
I saw this myself recently and ahd already fixed it :-) Code is now pushed to GitHub and Release V1.0.8.0 contains the binary for you.

Thanks for using GrblPanel

pflugj, 2016-02-04:
That was quick ;-)
Thanks

pflugj, 2016-02-04:
Sorry me again.

When I connect I get the same error again.
I can click on ignore and the program keeps running and the setting show 
now.

I also noted something else:
When I tried to enter some new numbers I noted the following:
The numbers (1-9) from the number-block (1-9) are recognized.
but the number "0" from the number-block is not. The "0" one on top of 
the letters works. My keyboard is OK! - strange!!

Thanks again for all your efforts ;-)
Jürgen

gerritv, 2016-02-04:
The last item is due to me using numpad 0 to set all axis to 0 :-( I will change that on the weekend.

What version of Grbl are you using? Have you changed the startup delay on Settings page? 

To ensure all the settings get reset, find the user.config file. Win 8/10 it is C:\Users\<yourusername>\AppData\Local\GrblPanel, Win 7 somewhere similar if I remember correctly.

pflugj, 2016-02-04:
Hi.

>
> The last item is due to me using numpad 0 to set all axis to 0 :-( I 
> will change that on the weekend.
ok ;-)

> What version of Grbl are you using?
Version 1.0.8.
>
> Have you changed the startup delay on Settings page?
No - currently shows  "1.5"
>
> To ensure all the settings get reset, find the user.config file. Win 
> 8/10 it is C:\Users<yourusername>\AppData\Local\GrblPanel, Win 7 
> somewhere similar if I remember correctly.
Here is the config file content:
<?xml version="1.0" encoding="utf-8"?>
<configuration>
     <userSettings>
         <GrblPanel.My.MySettings>
             <setting name="Port" serializeAs="String">
                 <value>COM3</value>
             </setting>
             <setting name="JoggingUnitsMetric" serializeAs="String">
                 <value>True</value>
             </setting>
             <setting name="JoggingFRImperial" serializeAs="String">
                 <value>100,50,10,5</value>
             </setting>
             <setting name="JoggingFIImperial" serializeAs="String">
                 <value>1,0.1,0.01,0.001</value>
             </setting>
             <setting name="JoggingFIMEtric" serializeAs="String">
                 <value>10,1,0.1,0.01 </value>
             </setting>
             <setting name="JoggingFRMetric" serializeAs="String">
                 <value>1000,500,100,5</value>
             </setting>
             <setting name="JoggingUseKeyboard" serializeAs="String">
                 <value>True</value>
             </setting>
             <setting name="JoggingFIDefault" serializeAs="String">
                 <value>I2</value>
             </setting>
             <setting name="JoggingFRDefault" serializeAs="String">
                 <value>F2</value>
             </setting>
         </GrblPanel.My.MySettings>
     </userSettings>
</configuration>

Thanks for your efforts - :-)

pflugj, 2016-02-04:
Sorry forgot.

I just updated GRBL to v 0.9j

What is the g-code command to verify the installed version?

- so far: the error did not pop up again -

gerritv, 2016-02-05:
The command is $I
I am testing with 0.9j as well.

gerritv, 2016-02-07:
Fixed the use of KeyPad 0 for setting all axis to zero, replaced it with key.A

closed 2016-02-07 by gerritv