gerritv/Grbl-Panel #46: Error reported on 1.0.9.0

Mac2712, 2016-02-07:
Just downloaded the new version 1.0.9.0 
When trying to Jog get the following error

************** Exception Text **************
System.ArgumentOutOfRangeException: Specified argument was out of the range of valid values.
Parameter name: value
   at System.Windows.Forms.DataGridView.set_FirstDisplayedScrollingRowIndex(Int32 value)
   at GrblPanel.GrblGui.GrblGcodeView.UpdateGCodeStatus(String stat, Int32 index)
   at GrblPanel.GrblGui.processLineEvent(String data)
   at GrblPanel.GrblIF.raiseAppSerialDataEvent(String data)
   at GrblPanel.GrblIF.VB$StateMachine_40__client_ComReadData.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.CompilerServices.AsyncMethodBuilderCore.<>c.<ThrowAsync>b__6_0(Object state)

went back to 1.0.8.0 all working ok

gerritv, 2016-02-07:
Fixed, sorry about that :-( Scrolling DataGridView's is interesting. See V1.0.9.0a file in the Release for 1.0.9.0

closed 2016-02-07 by gerritv

Mac2712, 2016-02-07:
Thanks for the fast response that's working great, love the fast file load.
Could I make a suggestion that the XML file for the macros stays the same file for each version.

gerritv, 2016-02-07:
Not sure what you mean by 'xml file stays the same'? Once you make a change to config, there is a file in Users/<yourname>/AppData/GrblPanel somewhere that should be your custom version. The .config file in the program folder is the original master and will update each release.

Mac2712, 2016-02-08:
Gerritv sorry for being a bit vague, its the file in Users//AppData/GrblPanel ever time you update to a new version it creates a new folder which contains a new blank file. you have to then make sure you update all the settings and macros on the new version (I missed the MM/Inch setting).

My question is can we not use a fixed location and file so all the data you have set is used by the new version ?

matmurdock, 2016-02-08:
Maybe it may be more useful to have an import / export settings feature
down the road.

Mat

gerritv, 2016-02-08:
I am looking at writing a custom SettingsProvider class to deal witht he problem. Might be a while though before I get to it. In meantime you will have to copy the user.config file from the prev directory to new one. :-(