gerritv/Grbl-Panel #48: Create custom settings file

gerritv, 2016-02-08:
GrblPanel uses vb.net, which saves Settings in specific locations. User.config is the file used to save your modified settings. This is stored in a directory based on user name, and application signature. As such it is a) not easy to find, b) the application signature changes for each release. The end result is that your custom settings 'disappear'.

The solution is to write a custom SettingsPRovider class. I will do this but not sure about the timeline. In meantime you will have to copy the user.config file from the previous directory to new one. :-(

On Win10 this directory is c:/users/<yourname>/AppData/GrblPanel

JasonTitcomb, 2016-02-27:
Try `My.Settings.Upgrade()` in your form load event.
[http://stackoverflow.com/questions/1702260/losing-vb-net-my-settings-with-each-new-clickonce-deployment-release](url)

gerritv, 2016-02-28:
@JasonTitcomb Thank you, 
I had been looking at a custom Settings provider but that seemed like too much work for what was needed. Will try this out soon

gerritv, 2016-08-20:
Closed in V1.0.9.5, Thanks @JasonTitcomb

closed 2016-08-20 by gerritv