gerritv/Grbl-Panel #49: Minor bug in UpdateGCodeStatus routine when datagrid is empty

joebananas10, 2016-02-10:
Tested out the 1.0.9a code tonight and kept running into unhandled exception errors when the datagrid was empty. 

For example. Open the program, connect to comport, press Unlock and then Home. After Home command finishes, change the cbxStateOffset from Metric to Inches. boom, unhandled exception.  Quit/close program and restart. Connect, unlock, home, click G55 offset set button. boom unhandled exception.

Running code in the debugger while connected to cnc machine revealed error was thrown right before the _dgview.Refresh() at the end of the UpdateGCodeStatus routine. Specifically, it was thrown on the IF statement in the last ELSE.
```
If (_dgview.FirstDisplayedScrollingRowIndex + _dgview.DisplayedRowCount(True) <= index) Then
    _dgview.FirstDisplayedScrollingRowIndex = index - _dgview.DisplayedRowCount(True) + 1
End If
``` 

I've corrected this by changing it to this,

```
If (_dgview.FirstDisplayedScrollingRowIndex + _dgview.DisplayedRowCount(True) <= index) And index <> 0 Then
    _dgview.FirstDisplayedScrollingRowIndex = index - _dgview.DisplayedRowCount(True) + 1
End If
```

Can anyone else test this and respond?

gerritv, 2016-02-10:
Thank you Ed.  I had more than a fewchallenges with those lines of code already :-(
The switch to using DataGridView has not been smooth, a complex object to work with. But worthwhile as it leads to MDI history, editing etc possibilities in future.
As soon as I figure out the performance problems I will push a new release.

gerritv, 2016-02-18:
Fixed in this release

closed 2016-02-18 by gerritv