gerritv/Grbl-Panel #50: Error reported on 1.0.9.1

Mac2712, 2016-02-18:
Gerritv I am sorry to report I am getting the following error when I connect the comm port

************** Exception Text **************
System.IndexOutOfRangeException: Index was outside the bounds of the array.
   at GrblPanel.GrblGui.showGrblPositions(String data)
   at GrblPanel.GrblIF.raiseAppSerialDataEvent(String data)
   at GrblPanel.GrblIF.VB$StateMachine_40__client_ComReadData.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.CompilerServices.AsyncMethodBuilderCore.<>c.<ThrowAsync>b__6_0(Object state)

gerritv, 2016-02-18:
Argh!!! It seems like the earth is moving under me. Is this right after you connect?
Can you dump your settings for me please? I use $10=31 with 0.9j. I do see an issue if this is $10=15 but that is exceptions in GrblStatus, where it deciphers the Q and Rx portions of the responses.

Mac2712, 2016-02-18:
gerritv I feel your pain but you where spot on I was using $10=17 with 0.9j. switched to 31 all ok. went back to test 15 and it works ok, back to 17 exception as soon as it feed back the change, tried going back to 1.0.8.0 and it crashes with 17 as well

The settings I was using
$0=10 (step pulse, usec)
$1=25 (step idle delay, msec)
$2=0 (step port invert mask:00000000)
$3=0 (dir port invert mask:00000000)
$4=0 (step enable invert, bool)
$5=0 (limit pins invert, bool)
$6=0 (probe pin invert, bool)
$10=31 (status report mask:00011111)
$11=0.010 (junction deviation, mm)
$12=0.002 (arc tolerance, mm)
$13=0 (report inches, bool)
$20=0 (soft limits, bool)
$21=1 (hard limits, bool)
$22=0 (homing cycle, bool)
$23=0 (homing dir invert mask:00000000)
$24=25.000 (homing feed, mm/min)
$25=500.000 (homing seek, mm/min)
$26=250 (homing debounce, msec)
$27=1.000 (homing pull-off, mm)
$100=400.000 (x, step/mm)
$101=400.000 (y, step/mm)
$102=400.000 (z, step/mm)
$110=2000.000 (x max rate, mm/min)
$111=2000.000 (y max rate, mm/min)
$112=2000.000 (z max rate, mm/min)
$120=100.000 (x accel, mm/sec^2)
$121=100.000 (y accel, mm/sec^2)
$122=100.000 (z accel, mm/sec^2)
$130=200.000 (x max travel, mm)
$131=200.000 (y max travel, mm)
$132=80.000 (z max travel, mm)

gerritv, 2016-02-18:
I have a fix coming so it works for 15, or 31 or 17 or whatever. Just in case, because until you can connect, you can't change the settings from within GrblPanel :-)

rwensley, 2016-02-18:
Gerrit,

Just a thought here.  How about a message box that says something like "grbl Panel has determined that the status port is not set to the proper value.  In order to run grbl Panel this parameter must be changed.  Would you like grbl Panel to change this value? Yes/No/Ignore"

Ron

rwensley, 2016-02-18:
Gerrit,

Release 1.0.9.1 ran perfect.  No issues what so ever.  Nice job thank you...

Ron

viewsat, 2016-02-18:
You may be able to refer 109JBs GRBL Interface, he said is Vsiual studio to write programs.

https://dl.orangedox.com/EqqZYCIOLOeHIrjLc6/109JBs%20GRBL%20Interface.zip

gerritv, 2016-02-19:
@rwensley I prefer protecting the code, pop ups are a nuisance in other ways. Plus I then need to ensure there is no race condition between starting Status polling and the changing of that parameter. And of course that parameter number might change in the future :-) 1.0.9.2 should be protected from this error (When I changed the serial IO code, vbCrLf was no longer getting dropped from the end of the lines on the way in to GrblPanel, thus the trailling '>' was being treated in conversion to numeric.

gerritv, 2016-02-19:
@rwensley Glad the performance is back.

closed 2016-02-19 by gerritv