gerritv/Grbl-Panel #51: GrblPanel-Release-1.0.9.1 resize max  send gcode slows down

viewsat, 2016-02-19:
GrblPanel-Release-1.0.9.1 resize max  send gcode slows down

see vedio>>  https://youtu.be/IWzbjGZSo0Q

gerritv, 2016-02-19:
I noticed the slow down as well. I even get different results depending on which of my 2 displays (laptop screen or second monitor) I have the GUI on.

In the next month or 2 there will be significant GUI changes coming (including fixed gcode display height) up so I won't be doing anything about this in the meantime. The stretch height was at the request of a few users early on but it doesn't fit in with my future gui design

In meantime please don't make the gui full screen height if your machine consumes gcode very fast or with lots of short moves. For those with slower cnc machines such as Bridgeports it won't matter either way.

closed 2016-02-19 by gerritv

viewsat, 2016-02-19:
Thank you, I will continue to test Grbl Panel's work.