gerritv/Grbl-Panel #52: About test V1.0.9.2

viewsat, 2016-02-20:
<1> I tested V1.0.9.2 send gcode no problem, but sent gcode time becomes very long.
I sent Roy Rogers Metric.nc this gcode normally about 5 hours 52 minutes, V1.0.9.2 more than 10 hours.

<2> If I change the fram vertical size, send gcode speeds vary greatly.
see vedio >> https://youtu.be/teW8dO3vfsM

gerritv, 2016-02-20:
Thank you, as I said for #51, I will not be fixing this soon. You can avoid the problem by not expanding the size of the initial gcode display.

I will look at adding a note on the wiki that GrblPanel is perhaps not best suited to use with laser type devices with a lot of small rapid movements.

viewsat, 2016-02-20:
May be add a function, the user can select ON / OFF gcode display.
Stop gcode display can accelerate send gcode speed?

gerritv, 2016-02-20:
I will make a version for you with that option. Sorry if I sounded sharp in the previous response.

gerritv, 2016-02-20:
Just to clarify "I sent Roy Rogers Metric.nc this gcode normally about 5 hours 52 minutes, V1.0.9.2 more than 10 hours.", are you comparing send time of the test version I posted to my personal website and 1092? Or 1070 and 1092?

What are your $$ settings for $100-132?

viewsat, 2016-02-20:
The Roy Rogers Metric.nc  I modified F=600 ,because my machine maximum speed F = 800mm/min
 
[image]

gerritv, 2016-02-20:
Thanks, can you also reply to my other question re: which version you are comparing 1092 to?

gerritv, 2016-02-20:
@viewsat try http://psgv.ca/Files/Release-1093viewsat.zip please.

Also, the extra 4 decimal places on the S lines are redundant and take time to send.

Until I implement buffer stuffing protocol (not soon!) I think this is as good as I can get balancing file load for huge files and send performance.

viewsat, 2016-02-21:
I Use Grbl-Panel V1.0.7 send Roy Rogers Metric.nc this gcode about 6 hours(about 100 lines / 2 seconds) . Change the size of the frame will not affect the speed send gcode.

gerritv, 2016-02-21:
Thanks, please try  http://psgv.ca/Files/Release-1093viewsat.zip

viewsat, 2016-02-21:
I email to you  attached a LON.cnc this gcode, there is no change in the S values, but still the same test results.

I try Release-1093viewsat.zip  ,
<1> The same test results and V1.0.9.2.
<2>Opened the Roy Rogers Metric.nc After that, if you want to open other gcode file you want to wait a long time.

viewsat, 2016-02-21:
V1.0.9a  ,V1.0.9.2, V1.0.9.3 ,test found a problem.
See vedio >>  https://youtu.be/I5e_fAnN9zs

gerritv, 2016-02-21:
"V1.0.9a ,V1.0.9.2, V1.0.9.3 ,test found a problem.
 See vedio >> https://youtu.be/I5e_fAnN9zs" I did ask you to not resize, knowing there is a problem?

(https://github.com/gerritv/Grbl-Panel/releases/tag/V1.0.9.3) no longer allows you to resize. I have also spent more hours on chasing performance, and am now at a point where doing more is taking too much of my time.
It could be that this is not the best program to feed your machine? Or you can alter it to suit your specific needs.

closed 2016-02-21 by gerritv

gerritv, 2016-02-21:
@viewsat, FYI the LON file you sent for testing took 0:32:06.28 to run with your Grbl settings.

rwensley, 2016-02-21:
Gerrit,

I ran a file that was 688890 lines long.  It loaded very fast and executed without error.  It took 11 hours to complete as it was made of mostly very small movements.  At no time did the machine stall or stutter.  No error codes were produced. All went well...except for my mounting which caused a waste of 11 hours and the material.

Nice job on the code.  Your work is appreciated.  Send me your paypal info and I will send you a donation.

Ron

gerritv, 2016-02-22:
Hi Ron, glad the program is working for you. Was the Q indicator getting any colour at all or just empty?

While I appreciate the offer of a donation, I can't accept it. I created this for my own use, which I will eventually get back to, and decided to share it on GitHub. The feedback from all the users who reported issues has been very helpful and often times educational.

rwensley, 2016-02-22:
The Q bar varied in fill with it being 100% most of the time. R bar would flash 1 or 2 segments from time to time but mostly blank.

I am poking around in the code trying to set the column width for the number in the dgv. Needs to show 10,000,000 lines for really big projects.

But I am a hack when comes to programming  so I will fiddle with it as I can.

Thank you for your work.  I appreciate it.
Ron

gerritv, 2016-02-22:
Check your email :-)

I will create a Feature Request issue for this item