gerritv/Grbl-Panel #54: Joystick

Encarnacao, 2016-03-05:
how would be possible to implement a joystick to move the cnc machine using grbl?
is there any tutorial in how to achive that?

wlaw, 2016-03-07:
Most senders like Grbl-Panel allow you to use the keyboard to move the spindle.
Using a program like keytojoy you can assign the keyboard movements to a usb joystick/gamepad.

I use it all of the time and it works well, this was one of the reasons for one of the upgrades.

once you install and test the joystick, install the keytojoy.

Setup your movements, Arrows, PGUP PGDown, +, -, *, /

Just a NOTE. I have to click on a distance for movement or somewhere in that box on grbl-panel to have control.

Good Luck.

gerritv, 2016-08-20:
@Encarnacao I updated the key mappings in Wiki. I use a joystick with http://www.xpadder.com

closed 2016-08-20 by gerritv