gerritv/Grbl-Panel #55: Macro Miss-behaving.

wlaw, 2016-03-20:
So I use this G-Code in a macro.
G90
G21
$H
M00
G92 Z0
G01 Z-20 F1000
F500
G38.2 Z-60
G92 Z0
G01 Z5
F50
G38.2 Z-7
G92 Z6.6
G01 Z30 F1500
G53 X-708 Y-1463.7 F4000
G92 X0 Y0

On the last line, the G92 dont set my X0 Y0.

On my very next move I receive a error Invalid gcode ID:24

If I load the G-Code as a file, it works great.

What do you think that could be.

gerritv, 2016-03-20:
Mnn, I know that I still need to implement some kind of 'wait for ok' on the macro stream, it presently uses the MDI mechanism which just hoses the commands to Grbl. Wonder if G53 access EEPROM (not mentioned in the list on the Grbl wiki).
Can you try adding an M0 after the G53? You will have to Start to continue.

wlaw, 2016-03-21:
I shortened the code and it seems fine. Instead of changing speeds so many
times. I use F2500 for both F1500 and F4000 mm per min.

Is it possible there is a limit to the length of the macro?

Thanks
Bill

gerritv, 2016-03-21:
Yes, that is a definite possibility and related to my not waiting for on Ok before sending out the next line. I will start looking at a solution now that the issue has really popped up.