gerritv/Grbl-Panel #59: Macros - What am I missing?

aeleus, 2016-06-25:
I'm running GRBL Panel 1.0.9.3 on Windows 10. On the Macros tab, I can create, save, and edit macros. What I don't see is a way to run the macros.

Here's a screenshot of the macros page:
[image]

gerritv, 2016-06-25:
You should see one or more buttons appear in the top middle section of the front page/screen. Click the appropriate one to run that macro.

aeleus, 2016-07-04:
I'm afraid that's not happening;

[image]

gerritv, 2016-07-04:
I will have to look at the code as to why, but I had to click Ok a couple of times for the buttons to show up.

gerritv, 2016-07-04:
I see 'End' button right below the MDI line on your screen shot?

aeleus, 2016-07-04:
Doh! How did I miss that?!

Thanks for your help and all the work you've put into GRBL Panel!

gerritv, 2016-07-04:
:-) It is smallish, and not blinking "Click me, Click me" :-)

closed 2016-07-04 by gerritv