gerritv/Grbl-Panel #60: Make the main window Resizable/Maximizable

intensite, 2016-07-15:
This is more of a rant than a bug.  Feel free to tell me off.

The window is currently fixed and can't be resized/maximized.

As this application in driving potentially dangerous machinery, I personally think it should be possible/desirable to have the window maximized while it is running.

The way it is right now is like a big dialog box.  I can accidentally move it off the screen and I am always afraid it will close if I hit the escape key.

Great app otherwise.

gerritv, 2016-07-15:
What do you suggest is the appropriate behavior of the various areas of the UI if it maximizes? A big border would be ok?
I could also disable the x in top right of screen, forcing user to go through the File/Exit path to close.
In the meantime I will do some experiments to see how it would look.

wlaw, 2016-07-15:
I think this is a great program and have said so many times. Hats of to you @gerritv. That said, it would be nice to have the function back to maximize the program. I remember that the speed of sending was affected by the size of the window.  However, the long running window of test is not so important to me. Yes I look at the commands that were sent, but most important to me is:
What line of code am I on?
What percent of the file is complete?
If sending stops, what line did it stop on?

[image]

In this way you could remove the long line of running commands, but show what is currently being processed.  Or possible use a Show / No Show button so the speed of the sending would not be affected. In any case, the only time I look at this window is to see how far the project is, and estimate the time to completion. Or, know where I need to cut my G-code to pick up where something went wrong.

As it is now, if I load a file with 100000 lines, I can’t tell what line I'm on, well it’s an educated guess.
Only because the line number column is too short and only displays 5 numbers.

Lastly, the macros are fabulous. I use them every day and they have changed the way I look at my cnc.
I have auto home with tool height offset, and more. 
Adding the macros to a different version is kind of troublesome. Would it be possible to just have the button point to a macro g-code file on the hard drive, load it then send it? That way, you would only have to copy the files into the new directory for Grbl Panel. 

Just some thought from my side, I love the program and use it all of the time. 
Great work!

Bill

gerritv, 2016-07-15:
Thanks both of you for the thanks.
I just this morning finished the 'ignore X for Close' code (can't remove it without also losing the Max/Min :-( )
Maximize is back as well.
Line number column will be resizable.
Copying macros to new version will be in next release, finally figured that out a week or so ago.

The issue with Gcode display wasn't sending speed as much as it was getting visually messed up. I will limit the display size of that grid even when Maximized.

I should have a large release available in a week, it includes a bunch of other enhancements as well in anticipation of Grbl 1.0.

I appreciate the occasional Issue being reported, I continue to develop the program even though my own CNC efforts have stalled for the past 2 years. I expect this winter to get back to my projects.

gerritv, 2016-07-17:
@wlaw  question about the Elapsed time. I have it mostly done but wondering at which point it should be cleared to 0? Should it be at the Start only as any other time (e.g. after an M30/Rewind) it would be nice to see the length of time that run took.
So the timer would reset to 0 on Load new file, Reload button (but no M30), Start.
I plan to update the time 1/sec during a run, that should be granular enough.

wlaw, 2016-07-17:
Sounds good. I think that reset on load file is good.

viewsat, 2016-07-22:
Hi gerritv,
My friend add & modify Grbl_panel some of the functional.

space key >> HOLD/RESUME
monitor >> on/off 
send >> send line number
modify frame size use 1024*768

Please receive email to download the file.

gerritv, 2016-07-22:
My email address is listed on this page: https://github.com/gerritv 

I will take a look and blend some of the items into the upcoming release. Send line number is already taken care of.

(I would have preferred a new Issue for this to keep things organized, please create  new one next time :-) )

viewsat, 2016-07-22:
Download the file my google drive >>
https://drive.google.com/file/d/0B0BGzc0LgeZRQW9NQ0JCanRaZkk/view

gerritv, 2016-07-22:
Downloaded, but no source code? No problem I will make some of the changes anyway,
Thanks

viewsat, 2016-07-22:
Sorry, my friend did not give me the source code.

gerritv, 2016-08-20:
See Release V1.0.9.5

closed 2016-08-20 by gerritv