gerritv/Grbl-Panel #63: Couldn't get V1.0.9.5 working on Win 10

CreateiveRobotics, 2016-08-26:
I just tried running the new V1.0.9.5 version on a windows 10 laptop. The application produced unhandled exception errors every time it encountered an unsupported command in the gcode file. When I clear the exception it just stalls so I couldn't get it to work at all and on one attempt I had to force the app to close using the task manager. I reverted to version 1.0.3.0 and ran the same file with no problems - it just reports the unsupported command and waits for me to prompt it to continue.

I'm running the latest version of grbl on an Arduino and controlling a shapeoko. Gcode is generated with CamBam.

gerritv, 2016-08-26:
Can you post a screenshot of the exception? and maybe a snippet from the file? I can't reproduce it. I am also on Win 10. 

BTW, you don't need to go all the way back to 1.0.3.0. All the releases are here: https://github.com/gerritv/Grbl-Panel/releases
A lot has been improved/fixed since that old version.

CreateiveRobotics, 2016-08-26:
I will try when I get a chance over the weekend. I reverted to 1.0.3.0 for purely pragmatic reasons - I was rushing to complete a job, it was already on my machine and it was the version I had been using up until then so I knew it worked with my setup ...

joebananas10, 2016-08-26:
I also had a similar problem getting V1.0.9.5 to run on Win10. The exception I was getting occurred during application startup. The unhandled exception would cause the entire program to close and I could never get it to load. I’ll get you a screenshot of as soon as I can. I was trying to come up with the resolution myself before I filed an issue. 
I have traced it down to the following block of code in GRBLSettgings.vb/FillSettings

```
            If (params.Count = 3) Then
                _paramTable.Rows.Add(params(0), params(1), params(2))
            Else
                ' We have Grbl in GUI mode, so add Description manually
                index = params(0).Remove(0, 1)
                _paramTable.Rows.Add(params(0), params(1), _settings(index))
            End If
```

The exception is thrown when index equals 14. I’ve added an IF statement to get past the issue for the moment because I haven’t quite figured out why it fails at only that moment. The new line I created looks like this

```
                If index <> 14 Then _paramTable.Rows.Add(params(0), params(1), _settings(index))
```

This allows me to run the app as normal but like I say, I don’t know what Index 14 means at the moment to know where the real problem lies.

gerritv, 2016-08-26:
Thank you @joebananas10 , that helps. I will have to test some more using both GUI and non-gui mode versions of Grbl.

I have now reproduced the fault, will release a new version tonight. Thanks all for your patience.

gerritv, 2016-08-26:
@joebananas10 what version of Grbl are you running? Your problem is actually different cause from @CreateiveRobotics although end result is the same.

I have fixed the original defect, when Grbl errors due to Unsupported Command (or any other error)

gerritv, 2016-08-27:
@joebananas10 , @CreateiveRobotics please try http://psgv.ca/Files/TestV1.0.9.6.zip . I fixed both defects as far as I can tell. You might also want to test using Enter key to change Grbl Settings as well as Offsets.

If this all works I will do a release!

joebananas10, 2016-08-27:
my grbl version is 0.9g.20140905. Specifically, my version comes from a Carbide 3D board for the Shapeoko3 machine. Lastly, v.1.0.9.6 works correctly, with no unhandled exception errors, for me.

CreateiveRobotics, 2016-08-27:
Just tried the 1.0.9.6 version without any problems so please do go ahead and release - great work, thanks.

closed 2016-08-27 by CreateiveRobotics