gerritv/Grbl-Panel #64: Conversational programming for basic tasks

MeJasonT, 2016-09-01:
Great work, love the GUI does exactly whats asked of it.  Thank you so much for spending your time and sharing your code.
I would love to see Conversational Programming tasks offered - perhaps from a drop down menu.
GRBL is just a command interpreter so its best concentrating on driving the machine.  jobs like pecking for holes and drilled hole matrices etc are easily produced with a string of Gcode commands.  i'm astonished that no one has worked on this area of machining for grbl yet.  I know its easy to produce it as part of your drawing from CAD and post processing using Cambam and the like but that means going to the drawing board every time for simple machining operations. Grbl also does not yet support drilling and pecking commands.

gerritv, 2016-09-01:
I have been looking at incorporating some of this code: https://pico-systems.com/gcode.html but need to do some rewiring of GrblPanel internals first. Planning for this for later this year. In meantime I recommend using those small utilities.

MeJasonT, 2017-08-05:
Hi Gerrit,
 i've been attempting to write a couple of canned cycles that would be GRBL (Uno) compliant,  It not as easy as i thought i would be.  It would appear its almost impossible to find machining routine that dont involve the use of Gcode commands outside of GRBLs command set.  I have manage to write a VB6 program to do drill pecking and i'm currently looking at Bolt hole/PCD.  Im abit of an old lazy engineer so my coding is limited and work ethic is getting slower, i'm like an old dog after so many beating i've given up.  I have attached the code from the VB form for laughs. This was the idea i had for grbl panel to have a tab with a dropdown list of canned functions that a user could quickly click on and generate code. this also ties in with my previous Issue/request for a feeds and speeds tool.

Which code do you use for GRBL-panel, or is simpler for you as a coder, VB.net of C#. i know just enough of both to be dangerous.  If my monstrous examples are of any use i'm happy to update you on my progress.

I have not tried it in anger yet but having viewed it in camotics it looks safe to run.
biggest pain is that it saves the files in the root user file C:\Users\(whoever)\  would be nice if it loaded straight into GRBL-Panel.

`Dim PlungeRate As Double
Dim Retract As Double
Dim TotalDepth As Double
Dim DepthPass As Double
Dim DepthToCut As Double
Dim AmountToCut As Double

Private Sub Command1_Click()

Dim i As Integer
Text9.Text = "G21 G90" & vbCrLf & "G0 Z" & Text8.Text & vbCrLf & "G17" & vbCrLf & "M3 S" & Text10.Text & vbCrLf
Text9.Text = Text9.Text & "G0  X" & Text4.Text & " Y" & Text5.Text & vbCrLf & "G0 Z" & Text1.Text & vbCrLf
AmountToCut = DepthToCut
PlungeRate = Val(Text2)
Retract = Val(Text6)
TotalDepth = Val(Text3)
DepthPass = Val(Text7)

DepthToCut = TotalDepth / DepthPass

AmountToCut = Val(Text1)

For i = 1 To DepthToCut
AmountToCut = AmountToCut - DepthPass

Text9.Text = Text9.Text & "G0  Z" & Format((Val(Text1) + Retract), "0.00") & vbCrLf
Text9.Text = Text9.Text & "G0  Z" & Format((AmountToCut + Retract), "0.00") & vbCrLf

Text9.Text = Text9.Text & "G1 Z" & Format(AmountToCut, "0.00") & " F" & Val(Text2) & vbCrLf
    
Next
Text9.Text = Text9.Text & "G0  Z" & Val(Text8) & vbCrLf & "M5" & vbCrLf

End Sub

Private Sub Command2_Click()
' Get a free file number
FileNum = FreeFile

' Create Test.txt
Open App.Path & "DrillPeck.nc" For Output As FileNum

' Write the contents of TextBox1 to Test.txt
Print #FileNum, Text9.Text

' Close the file
Close FileNum
End Sub

Private Sub Command3_Click()
Form2.Show
End Sub

Private Sub Command4_Click()
Formulas.Show
End Sub

Private Sub Form_Load()

End Sub
`
[image]

The output looks like this
G21 G90
G0 Z10
G17
M3 S800
G0  X100 Y50
G0 Z6
G0  Z6.25
G0  Z5.25
G1 Z5.00 F30
G0  Z6.25
G0  Z4.25
G1 Z4.00 F30
G0  Z6.25
G0  Z3.25
G1 Z3.00 F30
G0  Z6.25
G0  Z2.25
G1 Z2.00 F30
G0  Z6.25
G0  Z1.25
G1 Z1.00 F30
G0  Z6.25
G0  Z0.25
G1 Z0.00 F30
G0  Z10
M5

gerritv, 2017-08-05:
That looks pretty decent! I used vb.net because I wrote a lot of VBA for a few years. Prior to that it was C, Fortran, PL/I, Cobol etc sprinkled with assembler for Univac 9000 and 1100 series mainframes. Two years ago started with C# and loved it so will be re-doing GrblPanel at some time in that. And cleaning up some of the architectural mess I created with quick fixes and changes over time.

There are 2 quick soluions to what you are trying to do:
1. http://www.webmachinist.net/webmachinistorderpage.html has a special (today only?) for US30.
2. Adapt https://pico-systems.com/gcode.html to use GUI.

I use Visual Studio 2017 Community Edition for all my work, including for Arduino's and ESP8266's. It is free and very visual. Code completion, hints of parameters for functions, lots and lots of documentation (when stumped I do an internet search for what I am trying to do and examples pop up). Easy to set breakpoints and see variable values at time of stop etc.

For a better Save dialog, use Forms.OpenFileDialog, it is easy to set up with default directory and even file type.

The problem I created in GrblPanel is that I don't have a queue for MDI (and as a result also Macros) so there is no wait for Ok response after sending a command. This is why Macros are somewhat limited in this program. I have twice tried adding a second queue (along the one with actual gcode program) and ran into various troubles.

The Feeds/Speeds thing got very complex very quickly. I think for me as a rank amateur machinist (who does lathe and mill work 'by feel and sound') those are only broad guidelines. I often don't know the actual steel type etc. Were I doing any CNC work it would be different but I haven't had a working machine since 2014! :-( Maybe this winter......

MeJasonT, 2017-08-05:
Hi Gerrit, Thanks for the reply.
Instead of Macros i thought it might be easier to simply populate the imported code window in the gui with the results from a popup. that way it would appear in the code window (i imagine you used a list control).  The advantage would be that once the file is set to run the code would already be queued.   I have had a look at the links, one you sent me before.  They are fine for an offline tool but not as magical as having them at your fingertips in the gui.  The webmachinist one i am also familiar with but if i remember right it produces some g commands not supported by grbl.  its almost like a conspiracy on the web preventing people from getting middle school educational material so the code comes from basic concepts.  Interesting history, mine is not so intense, machine code Z80, motorolla 68000, Pascal, visual basic, some QNX and thanks to arduino C#.  PLC's Allen Bradley and Siemens and now good old NC.  My career has had more of a mechanical and practical aspect to it, the coding has just been a skill picked up along the way - all be it a limited skill.

Im stuck wondering how you manage to keep Grbl-panel up to date with no machine to test on.

gerritv, 2017-08-05:
I hadn't thought of populating the gcode list directly as if it were a physical file. That would be quite easy to do.
The code examples were to reduce the thinking effort to come out with the gcode for those operations. I will tinker with a gui for 2 of them to see how easy it is to add.

The first 5 years or so of my work at Univac was mostly mechanical as well, card punches, tape drives, printers etc. Then shifted mostly to troubleshooting to tube/transistor/chip level.  I did a stint designing pcb's for a BMC IF800 Z80 machine. Also worked on networking code on a 68k Heurikon board (still have 3, do you need one :-) AES Data (one of the first word processor companies) tried switching to QNX in the late 80's but they were past saving so folded.

Doing the gui work without a machine is fine except for things such as door open and limits. But those are Grbl functions, I just need to produce the correct commands, send them and deal with the responses. But it would be more rewarding for me personally to have a working machine. In the meantime I do it for the love of it.

MeJasonT, 2017-10-11:
Hi Gerrit 
Not heard anything from you for a while, hope all is well.
Finally managed to get the latest Grbl and Panel running on both my Plasma table and milling machine.
I have been quietly working away on this enhancement in the background but to be honest my work has been unfruitful.  I still cant get my stupid head around feeds & speeds or  Drill hole circle (PCD).  Its so nice to see others jumping in with suggestions - that's an invitation by the way.
Its probably that only the dinosaurs are still using Arduino's and Grbl Panel and the others have jumped ship for TintG etc.
The Panel Gui is working well, no hiccups and is producing some great results.  Thanks again for you great work, still very appreciated.
Jason