gerritv/Grbl-Panel #65: Coordinate Measuring Machine

MeJasonT, 2016-09-01:
Would it be possible to implement coordinate measurement like the add on for Mach3 to reverse engineer ready made parts, ie. turn a current CNC machine running with Grbl and Grbl Panel to also work as a CMM Machine.  Mach3 has the distinct disadvantage of running on an obsolete parallel port on most modern PCs and laptops, the conversion for Mach3 to use USB and modbus is laborious to say the least.
Unfortunately Im pants at coding but can see how it may be possible using a bed levelling function and recording the coordinates when the probe touches.

gerritv, 2016-09-01:
I'm sure it is possible but not something I would be interested in adding any time soon.

closed 2016-09-01 by gerritv

MeJasonT, 2017-01-19:
Found Open Lic. CMM probing function
Found Toe's GrblGru, there doesn't appear to be a wiki or git site for it but it can be downloaded here [http://www.myhobby-cnc.de/forum/thread.php?board=5&thread=3&page=3](url) found through shapeoko forum. not tried it yet.

gerritv, 2017-01-20:
GrblGru looks interesting but not much info in English although my German is passable. No source, no real documentation and seems more geared to being a CAD/CAM integration. So really not use how that is of use to GrblPanel?