gerritv/Grbl-Panel #68: Spindle override on off cause error

tklus, 2016-10-18:
When I pause the program and turn the spindle off via the spindle override button it gives me an error. The error reads "Fatal error on write to Grbl, the I/O operation has been aborted because of either a thread exit or an application request"

Here is what I do to get the error. 
1. load a new program. I was running a program that circular interpolates a bunch of holes.
2. start the program. the machine starts running like it should.
3. at one point, I wanted to stop the machine and turn the spindle off so I could go in the house and not leave the machine running unattended. I hit the slide hold or pause button on my machine.
4. I press the [spindle] button that is on the right side of the GUI in the overrides area of the GUI. This turns the spindle off.
5. I then press the [spindle] button again to restart the spindle. the spindle starts back up
6. then hit start and the machine continues the program. after a few seconds the machine will give the alarm.
7. I found that several times I could start the program and hit slide hold right away and toggle the spindle on and off and the program would continue without the error.

Let me know if you need anymore info. this is not a big deal as I will just not bank on using the spindle override to turn the spindle off mid run. just figured you should know about the issue.

Thanks!
Tim

gerritv, 2016-10-18:
Hi
Can you try the same process using the Hold on GrblPanel instead of Hold on your machine?
To clarify #6, did you use Resume on Grbl Panel or something else?
Which Alarm number did you get or do you mean the Fatal error pop up?

 Just trying to get the exact sequence so that I can reproduce, otherwise difficult to trace and fix...

tklus, 2016-10-18:
#6 gives the alarm that I show in the image. I hit the cycle start button that is wired up to my machine. I will try pausing the program within grbl panel and see if that works better.

tklus, 2016-10-19:
I tried initiating a pause through grbl panel and the spindle override does not seem to work when paused by the panel. the machine stops motion but the spindle override does not change anything. the numbers in the display for spindle speed change but the change will not go into effect until I continue the program.

The spindle speed adjust override does seem to work when the program is running as well as feed rate override, which is great! I just can't turn the spindle off completely while running. in the scheme of things, this is not a big issue and probably not worth your time to fix. just figured folks should know about it.

Thanks for all the work you have done with this!

Tim

chamnit, 2016-10-19:
@tklus: Spindle stop overrides will only work while in the Hold state. This is to prevent it from stopping while in motion. Spindle speed overrides have a lower limit of 50% by default so you will not be able to turn it off while running. Only slow it down. 

You may have uncovered an implementation issue in Grbl. You should be able to alter spindle speed while in Hold. I did not think I accounted for this scenario. I'll check it and issue a fix if I find if it doesn't do this.

gerritv, 2016-12-12:
I think this is fixed in Grbl now so closing it.

closed 2016-12-12 by gerritv