gerritv/Grbl-Panel #70: Feature request - Scalable UI

CreateiveRobotics, 2016-10-23:
This is a feature request, and it may be impossible or far to difficult to implement but ... I've just started using this on my touchscreen laptop. Being able to interact without using the mouse or touchpad when machining is great but the buttons are a little small. Is there any way to scale the entire app to fit an HD screen (1280x720 or 1920x1080) so that all the controls, in particular the radio buttons, are slightly larger ... ?
Failing that, could the locations of UI elements be scaled relative to the window size so that you can maximise the window and have the UI elements spread across the full screen - this would help fat fingered people like me hit the correct control more easily.

tklus, 2016-10-23:
I also use a touch screen and that would be a great feature. Maybe there could be 2 text and button size settings, small and large

wind-rider, 2017-01-15:
@gerritv I'm interested in making an UWP app. The XAML UI elements seem to be better targeted than Windows Forms elements - is this something you'd be interested in, too?

Another nice advantage of creating an UWP app is that you can run it on a Raspberry Pi (with Windows 10 IoT Core) for example dedicated to run your CNC machine.

Today I started writing a connector library for Grbl 1.1 (early work at https://github.com/wind-rider/grbl-connector; I took inspiration from https://github.com/dotMorten/NmeaParser). This could be something that could be used

gerritv, 2017-01-17:
@wind-rider I too have been looking at XAML but not decided yet. Xamarin would provide for more platforms but its forms are limited to the lowest common denominator. A bigger risk is losing Windows 7 for those many using it.
I just don't have a lot of time right now to learn all this. It reminds me of when C++ first came out, I expect my first use to a be major fail :-(

wind-rider, 2017-01-17:
Thank you for your feedback.

I think supporting Windows 7 or not is a discussion on its own (people could upgrade to Windows 10 for free and it is not supported anymore according to https://support.microsoft.com/nl-nl/help/13853/windows-lifecycle-fact-sheet).

As for Xamarin - I couldn't get from your comment whether you know this already - using portable code and Xamarin it is possible to support Android, iOS, MacOS, but XAML for Xamarin and XAML for UWP applications is not per say the same. With UWP you can target windows desktop, windows mobile, raspberry pi (and Xbox and hololens, but I don't think these are useful targets for grbl).

As for learning - I can imagine that you fear a learning curve and that it will take quite some time to get to the same feature level as you already have. I think this choice is something you can judge best for yourself. Personally I don't consider myself an expert too, but I guess I can make something working. I agree about it taking time to get to the same feature level; however if you want to make steps forward w.r.t. the user interface I think Windows forms is not a future-proof direction.

gerritv, 2017-01-17:
I have no fear of learning, it is just time consuming. I wrote my first program in NCR315 assembler in 1965, lots of languages and machines later I am still happy to learn :-)
First steps regardless of the technology is to story board the UI. Once that is done I will think about architecture and implementation. GrblPanel arrived at its present state without a lot of design, it is now showing a strain as I want to add functionality.

I will watch your project in the meantime.

wind-rider, 2017-01-17:
Sorry if I expressed myself not well :) it is really cool what you've already made! And I understand that a learning curve takes time.

It is a good idea to play with ideas for the UI. I think there are parts that work well and others (I didn't expect having to doubleclick in input boxes in the G54-G59 editor) that could be improved.

I could help making a mockup in code, but personally I'm not the best designer..

gerritv, 2017-01-17:
The double click was fixed a couple of Releases ago by adding Return key to the same boxes and a few others on that screen. It took some time to figure out the solution..

Jimboe98, 2017-05-17:
A fully scale-able interface may well prove to be near impossible to achieve. However a solution may be to just use common resolutions. For example, 1024 by 600, 1024 by 768 , 1280 by 720 and 1920 by 1080. This would just mean designing forms to suit the resolution.