gerritv/Grbl-Panel #72: The program stops work with an error

AutoRich, 2016-10-30:
Hi!
The program stops work with an error.
I use GRBL 0.9j.
Failure happened in versions of the program:
GrblPanel-Release-1.0.9.3
and
GrblPanel-Release-1.0.9.7
Help to solve this problem please.

[image]

Complete description of an error in the file:
[Error-1.txt](https://github.com/gerritv/Grbl-Panel/files/560319/Error-1.txt)

gerritv, 2016-10-30:
Is Status Report Mask $10 set to 31?

AutoRich, 2016-10-31:
It was succeeded to find out a cause of error.
At the beginning of 10$ =3 - the error appeared.
Then set $10 in 17 and 31 - the error appeared.
When set $10 in 0 - the error ceased to appear!!!
But now in the program coordinates are not displayed.
[image]

gerritv, 2016-10-31:
There is no Try/Catch in that subroutine.
Can you screen shot with Verbose on, I need to see what Grbl is sending. There is something not correct with the message being parsed. I expect to see both Mpos: and Wpos: on Grbl 0.9

AutoRich, 2016-11-02:
Hi!
Here chtovyzyvat an error.
[image]

And here a code which was sent to the machine.
[image]

gerritv, 2016-11-02:
I am going to guess that you have an electrical noise problem with all those M3/M5 pairs. The last line shown has an extra ':' in the sequence 62.703,3:62 as well as the pattern of numbers being quite different from those above it.
I have tested with 0.9j and $10=3 with no problem. I gather that the file runs for a while before giving the error?

AutoRich, 2016-11-02:
At $10 =17 the error jumped out at once at connection.

gerritv, 2016-11-02:
That is because I expect both Mpos and Wpos to be present, thus the 2 low order bits of $10 must be set, e.g. 3 or 31. Otherwise I would get complaints that the program is not showing both :-)

electrokean, 2016-11-03:
Definitely looks like some data getting lost in the communications stream.
You're not using an Arduino with a CH340 serial to USB chip are you?

AutoRich, 2016-11-05:
Yes, I have Arduino with a CH340.
Programmed Arduino on new, changed an USB cable - fulfilled the first time without errors, and the second time the error appeared. (10$ =3)

gerritv, 2016-11-06:
The data corruption can be due to many things. 
E.g. PC gets busy and can't keep up with data coming from Grbl or electrical noise or ... There is unfortunately not much I can do in Grbl Panel to help with figuring that out. I can't protect everything with Try/Catch to avoid an Exception.

AutoRich, 2016-11-06:
If to you it is not difficult, make please in Grbl Panel an opportunity to trace this problem. On a power supply of problems should not be - I have a powerful power supply unit - 12V, 20A.
Still tried other programs - there did not notice problems.
But from all programs that I tried - your the BEST!!!
Thanks!

gerritv, 2016-11-11:
Please try http://psgv.ca/Files/Debug.zip

AutoRich, 2016-11-14:
Hi!
Today tried to start the program and here what error appeared almost right at the beginning.
GRBL v1.1d firmware
Parameter is 10$ =3
[image]
[image]
[image]
[Error.txt](https://github.com/gerritv/Grbl-Panel/files/589033/Error.txt)

gerritv, 2016-11-14:
I fixed/worked around it based on your using 0.9j !
I will do another for you for 1.1D, the internal code is quite different. You still have a serious problem somewhere :-(

AutoRich, 2016-11-14:
Hi!
Thanks for the help in a solution of the arisen problem.
No, more problems are not present. I decided to try 1.1d - thought can on this firmware of problems not will, but you saw result.
At 10$ =0 works well, did not notice any failures.

gerritv, 2016-11-14:
I want to be clear, I am not actually fixing your problem, only letting things progress in spite of it. Whatever your PC/machine combination is, it is losing some data on the way in to GrblPanel. That is why there is an error, the message is corrupted. $10=0 also hides the problem, not fixes it.
The last line of the message above has 1.053,0.0s where the 0s is clearly showing that WPo is missing.

gerritv, 2016-11-15:
Please try http://psgv.ca/Files/Debug.zip . To repeat, this does nothing to fix your actual problem, it only lets the message get past checks. This work around will not be in any actual release.

AutoRich, 2016-11-15:
Hi!
Made testing - the error jumped out:
[image]
[Error.txt](https://github.com/gerritv/Grbl-Panel/files/591773/Error.txt)

gerritv, 2016-11-15:
Ok, you need to find out why your setup is losing so much data. In the last line above there are 2 Ov: sections and the first one is garbled. I don't see how me modifying my program to accept these incorrect messages is helping. If I accept garbage then you still have a problem even if it looks like it runs ok.
What is your PC, what is the OS (Windows 7/8/10?)
When you run another GUI and look closely at the responses are there errors in those messages as well?

electrokean, 2016-11-15:
@AutoRich have you replaced your Arduino with one that doesn't use a CH340?
If so, what type are you using and what chip does it use for USB to serial?
Also, as @gerritv mentions, please provide info on your PC (CPU speed, RAM, OS, etc).
One of these is going to be the cause of this corrupt communications, not grbl or grbl-panel.

gerritv, 2016-12-02:
Closing this as no new news.

closed 2016-12-02 by gerritv