gerritv/Grbl-Panel #73: Access to the port 'COM3' is denied.

Darathy, 2016-10-30:
Today i decided to cut something with my CNC and i get this message:

Access to the port 'COM3' is denied.

In the details it says:
************** Exception Text **************
System.UnauthorizedAccessException: Access to the port 'COM3' is denied.
   at System.IO.Ports.InternalResources.WinIOError(Int32 errorCode, String str)
   at System.IO.Ports.SerialStream..ctor(String portName, Int32 baudRate, Parity parity, Int32 dataBits, StopBits stopBits, Int32 readTimeout, Int32 writeTimeout, Handshake handshake, Boolean dtrEnable, Boolean rtsEnable, Boolean discardNull, Byte parityReplace)
   at System.IO.Ports.SerialPort.Open()
   at GrblPanel.GrblIF.Connect(ConnectionType typeIn)
   at GrblPanel.GrblGui.btnConnDisconnect_Click(Object sender, EventArgs e)
   at System.Windows.Forms.Control.OnClick(EventArgs e)
   at System.Windows.Forms.Button.OnClick(EventArgs e)
   at System.Windows.Forms.Button.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

I have tried runing Grbl Panel as administrator with the same message. I am using xPro controll board for my CNC and using Grbl panel 1.0.9.7. 1.0.9.5 with same problem.

Anyone know how i can fix this?
[image]

gerritv, 2016-10-30:
I am reopening this as there should be better handling. I assume another program had grabbed the Com port?

Darathy, 2016-10-30:
It fixed itself after i restarted the PC dont know why it got blocked in the first place.

closed 2017-01-17 by gerritv