gerritv/Grbl-Panel #78: Error on connect

Mitchell-Hampton, 2016-12-12:
I am getting a unhanded exception every time i connect to my grbl. 
[image] i can continue but it is pointless as all data seems to be garbage. I tried debugging it and it seems to point to GrblPosition.vb line 97 as workPos is only 2 long and it is trying to access the third member. I tried fixing quickly with no understanding of the system by adding some checks which got rid of the error popup but my data was meaningless. 
[image]

Mitchell-Hampton, 2016-12-12:
I will try this on my other computer tomorrow to test if its something weird on my PC as I assume its not common at all since it completely inhibits functionality

BernardG, 2016-12-12:
Thanks for reporting this, Mitchell. What is the exact version of Grbl-Panel you are using? This ring a bell, I saw this while localizing, and believe to have fixed it! If you are using release V1.0.9.8, then I'll look at this, as I may have created a bug! (Or you may have to use Windows in French, as it works for me here! LOL)

gerritv, 2016-12-12:
The cause seems to be that GrblPanel is not recognizing your Grbl as being 1.x version so it is executing the Grbl 0.9 protocol code. E.g. you should see the Override section of display on RH bottom of GrblPanel as well as the pin status'
What version of Grbl are you running? Is it a custom version?

Mitchell-Hampton, 2016-12-12:
I have tried both V1.0.9.8 and V1.0.9.7. I am running Grbl1.1e.

gerritv, 2016-12-12:
Very puzzling then. 
Can you show what is in Status panel around the time that Grbl ID string appears?
If you click Details on the Exception popup, what line does it indicate the error is detected on?

Mitchell-Hampton, 2016-12-12:
I have just run V1.0.9.7 on my other computer and a new Arduino and it works fine. Is it possible that it its caused by my USB-Serial adapter that I use on the other system?

gerritv, 2016-12-12:
There are some that cause problems, CH340 is known to have issues at 115200 bd. 
Are you ok with closing this issue then?

Mitchell-Hampton, 2016-12-12:
Yea it is OK with me, Thank you this whole Grbl thing is amazing :).

closed 2016-12-12 by gerritv