gerritv/Grbl-Panel #79: tool length offset G43.1

Mac2712, 2016-12-13:
In the offset sub tab there is a tool length offset G43.1 value box, but there is no means to send a value that you have changed, you can only set it to zero

gerritv, 2016-12-13:
Doh, I didn't add code to set that :-(. It would seem you are the first to use this. Workaround is f course to send G43.1 z<value> from MDI.
Fix is tested, release soon.

Mac2712, 2016-12-13:
MDI is what I am using. its great for phantom runs Just haven't sent you a bug report.

As your getting closer to a very polished piece of software, I will start writing down some of my thoughts. I have assess to professional CNC kit so have more exposure to see the good and the bad.

gerritv, 2016-12-15:
Fixed, thank you for finding this.

closed 2016-12-15 by gerritv