gerritv/Grbl-Panel #80: Enhancements

Mac2712, 2016-12-13:
gerritv can I suggest some enhancements

Can there be somewhere to read the version of GRBL possibly the about box

A large indicator that there is a override active. suggest a red outline around the override section 
I tripped up on this started a new part at 200% override.

A larger box with more lines for editing macros also the last line in the box shows no text even when theirs more lines of code.

Macros can we not save these so when we upgrade the software, they remain

Hope these give you some ideas
Been using grbl + panel now for 6 months and IMHO still the best

gerritv, 2016-12-13:
Good points:
1. I will add handling of OPT for Grbl 1.1, and also display of Grbl Version to probably a new page.

2. I will look at this, easy for the 3 speed overrides. The Spindle/Cooland and Mist overrides are more problematic as Grbl doesn't actually feed back what the override status if for those 3. It shows whether S, F or M is active but those indicators are also set if you send e.g. an M3 or M8 :-( I will need to add State for those 3 items, and then figure out in what other cases (e.g. Rest) besides a button press to clear the state.

3. This was implemented some  time ago, is it no longer working?

4. Single Step has been on the list for quite a while, will have to look at sample code someone sent me.

Glad it is working well for you. I am looking at a major revision next year. What OS are you using, would you be cut off if we went with Windows 8/8.1/10 only (UML app)?

Mac2712, 2016-12-13:
gerritv
1 it would be good, as there is a lot of upgrading going on !
2 just noticed spindle/coolant showing override when options set from gcode 
3 agreed but open up box length next release please 
4 not working here how is it meant to work?
5 Single step and go line even better  

I am using Win10 on a touch screen laptop works well

gerritv, 2016-12-14:
3. Is done, will release soon

4. When the version number of GrblPanel changes, the next time you run it detects that it needs to 'upgrade' the settings. E.g. I incremented my local copy of code to 1.0.9.10 and it tripped the code, copying the settings for macros etc across.

gerritv, 2016-12-15:
All done, hopefully the copy of Macros to this version will work for you.

closed 2016-12-15 by gerritv