gerritv/Grbl-Panel #81: Retrieving macros @ new revision - V1.0.9.10

Mac2712, 2016-12-15:
Gerritv
Just updated to new version 
 The macros from the old version are not being imported to the new version as previously discussed
Do you have to have to use the same directory when you upgrade ?

As you don't use an installer what is the correct installation procedure ?

I download the zip file extract it to a folder (GrblPanel-Release-1.0.9.10) then run it from there, via a shortcut on the desktop.

on the plus side all other changes work fine.

gerritv, 2016-12-15:
The user.config file is in this tree: C:\Users\<yourname>\AppData\Local\GrblPanel. There should be a dir for each version. You can manually copy the user.config file.

Can you let me know what the tree is and if the previous version is there. And if that one contains the macros?

Mac2712, 2016-12-16:
Gerritv,
Under the GrblPanel dir the is about 10 dir with names like "GrblPanel.exe_Url_zhhfidyomewrbysaxbrcvhcdhiig5wic" then under that theirs the config file.

Did the earlier versions have a default probing macro ? I ask, because recently theirs none by default

gerritv, 2016-12-17:
I think you are confusing the directory naming algorithm by installing GrblPanel in different directories. If you install to same dir each time, it should all go into the one place in AppData. 
You can fake it by copying the previous version dir into the new directory, delete the current version directory (in this case 1.0.9.10) and then run 1.0.9.10 again. This should copy your settings etc across.

There was a default macro but I took it out a while ago. Wasn't sure it was useful.

I will start looking at an actual installer again (Wix) and a separate xml file for macros.

Mac2712, 2016-12-17:
Ok I deleted all the tree under \AppData\Local\GrblPanel installed 1.0.9.9 ran it and added a macro restarted to check it was there all OK copied 1.0.9.10 to the same Dir and started macro did not appear. started it again and it appeared on the second start

I also noted that there is in GrblPanel.exe.config a macro in the first slot <value>Probe,G38.2 Z-30 F10</value> but when this is changed from Probe, to Probe: like the other macros. it apperes on the MDI as a button but you can see it in the Macro tab unless you run it a second time.

So from my testing you must install into the same dir as previous ver
also if you want to leave in the probing macro the "," needs changed into a ":" but the macro is not that good so you may want to remove it or replace with a better one like below 
G92 Z0
G38.3 Z-30 F10
G10 P1 L20 Z0.5
G0 Z5

And you must run the program twice ?

gerritv, 2016-12-17:
Glad it is working, sort of, for you. I think it will be quicker and more beneficial to implement an installer along with separate macro file than figuring out the automatic update stuff.  E.g. I will not have a solution for the changed install directory regardless of how much time I spend on the present update mechanism. It is quite opaque :-(

Mac2712, 2016-12-17:
I totally agree with a Installer people will not have to think about where it should go