gerritv/Grbl-Panel #84: Keyboard key assignments

MeJasonT, 2017-01-02:
Would it be possible to have:- reset, hold, start and unlock (pendant functions) made available on the keyboard.  Iv'e come up with a pendant idea using Antimicro joystick to keyboard application 
[(https://github.com/AntiMicro/antimicro/releases)] and  a Gamepad as a pendant.  works great for jogging with the latest release but not able to start/stop the machining program.  This would enable users to place their PC/Laptop in an RF free environment and simply use the pendant to interact with the program.
my intention would be to pull the game controller apart and 3D print a new shell which would look and feel more like a pro pendant.

Interested to hear your thoughts.

gerritv, 2017-01-02:
Certainly. Do you have suggested key mappings for those controls?

jahnj0584, 2017-01-02:
Ctrl+H,CTRL+R and CTRL Spacebar

gerritv, 2017-01-02:
Start/Resume is already toggled with Space so I will only need to add Ctl H (Hold), Ctl R (Reset) and Ctl U (Unlock)

MeJasonT, 2017-01-02:
Thank you very much gents, you have made my Christmas.
I like the keymap suggestions that have been suggested and would quite happily work with them.
how is this for an idea, as the GRBL controller would not require the Pendant pins anymore as they could be controlled via the GUI this could free up pins for a fourth axis.  For safety reasons the reset would need to stay on the GRBL hardware to instantly stop the machine, as its the the reset pin for the micro controller anyways this would make sense.  I don't even know why I would want a fourth axis.

gerritv, 2017-01-02:
Glad you had a good Christmas :-)
Yes, it would free up a few pins but 4th Axis won't be coming to Grbl 1.0 (not officially) nor GrblPanel. When the ARM version of Grbl comes out later in 2017 I will have a version of GrblPanel to go with it. See http://github.com/gnea/grbl Issues for various discussions on what Gnea's features might be.

gerritv, 2017-01-17:
Done, plus wiki is updated. Thank you for the suggestions

tklus, 2017-01-17:
Hey MeJasonT, do you have any videos of your setup? I am interested in doing a pendant like this as well. Looks like a good idea. Do you have a way of mapping the increment selection on the joystick as well?

tklus, 2017-01-17:
now that I see the key mapping in the Wiki it looks like this has all been done! Can't wait to try this out!

MeJasonT, 2017-01-19:
Hi Gerritv,
Is it me or is the "/" key not coded - cant increase the feedrate only decrease "*".
Tested using keyboard and AntiMicro

wlaw, 2017-01-20:
I use this all of the time on my USB Game controller.
I use joytokey,

gerritv, 2017-01-20:
It works fine with the / and * on the keypad on my laptop.

MeJasonT, 2017-01-21:
OK its me, time to go and scratch my head

Mac2712, 2017-01-21:
I was thinking of creating a pendent with a Arduino Pro Micro which can be used as a keypad
But I did some testing of the key mappings 
This was my results
I noted a difference between some keyboards and laptops /,* for instance are part of some laptops Numpad 

[image]

Unfortunately I could not get the ones in red to work also rapid override only goes down to 50% and not 25%. 

I would appreciate some input Gerritv

gerritv, 2017-01-21:
Were you testing from a physical keyboard or from your controller?

tklus, 2017-01-21:
I was curious if you can assign keyboard strokes to macros. That would be nice for my laser machine as I turn the laser on to locate my 0,0 location. So I would use M03 S1 to turn the laser on at low power.

gerritv, 2017-01-21:
@tklus Macros hold gcode etc only. Whatever is in the macro text goes to Grbl.

Mac2712, 2017-01-21:
@gerritv keyboard only
On the ones using the Numpad have you tied the output of the pad or the actual key
Trying to map buttons on a Arduino to keyboard but failing on the numpad buttons

gerritv, 2017-01-21:
The code is not explicitly looking at numpad or otherwise. I will dig out my Grbl board and check tomorrow to see what is going on.

gerritv, 2017-01-22:
I think I have it sorted out. The Alt modifier won't be used, too much conflict with its default usage.
So: 
-Ctrl R becomes Ctrl X for Grbl Reset
- Alt R becomes Ctrl R for Reset Rapid Override
- Alt S becomes Ctrl S
- Alt F becomes Ctl F

I need to do more testing, should have a release late Sunday.

gerritv, 2017-01-22:
@Mac2712 Please test .12 to see if I corrected the issues you found. My testing works. The one potential issue is with the non-numpad /, it seems at least on my keyboard (HP Probook 4520S) that it is an OEM key.

Mac2712, 2017-01-22:
@gerritv 
Ok first problem when you enable keyboard arrows this stops most of the enhanced keyboard mapping,
switch it back off then the arrow keys don't work ?

This took me a while to work out what's was going on
Still testing some more stuff but that stumped me as I had tested some keys then went back after and none would work  👎

Mac2712, 2017-01-22:
Ok tested all
[image]

The ones in red don't work on my keyboard
can you map additional keys to the same button what about using the F keys ?

From my pendant arduino also having problems with mapping Numpad + and -  also these dont work from the main keyboard while * and / do.
yet the results on an text editor looks the same ?

gerritv, 2017-01-23:
@Mac2712 please try this version. If there are still keys that do not work I need the codes from the utility whose link I posted earlier. There are a bunch of OEM keys which come into play.
[TestRelease.zip](https://github.com/gerritv/Grbl-Panel/files/722591/TestRelease.zip)

Mac2712, 2017-01-23:
# Yes
@gerritv I have managed to test and it all seems to work
I even managed to get the hex codes for keypad + and - which where 0xDE and 0xDF

Ok can I ask if you could add cycle start and step that would be the nearly the ultimate 
But if we could add macro 1 as a key then  would use it to probe or whatever the code in the macro did??
We would have a winner

Mac2712, 2017-01-23:
And if anyone is intrested here are the codes from an Arduino Leonardo

[image]

MeJasonT, 2017-01-27:
Sorry chaps, I know I started this thread - I just got busy for a while.
I have an elderly mother-in-law giving us the run around so unfortunately have not been at my PC for a week or so. Good progress has been made in my absence and I will get on and test the functions at my end this weekend.

MeJasonT, 2017-01-27:
tklus, im wanting to share my pendant idea with you and the nice peeps on this site but don't wish to go off on a tangent on this thread/issue.  when i get it written up i will issue it as a separate thread for everyone.
cheers
J.

Ps. Gerritv where is your paypal donate button for this grbl project.

gerritv, 2017-01-27:
@MeJasonT You could open a Github project for yourself? I guarantee fame and glory :-) I would reference your project from the wiki here. With a bit of forethought your pendant would work with other Grbl GUI's.

Re: Paypal button, I am retired.  Being retired I don't want the explicit and implied commitment of accepting donations.
I started this to make use of a CNC machine  I built in 2014. Strangely I haven't actually used CNC since then but enjoy keeping GrblPanel alive. I do have plans for later this year to get going again with CNC so a redesign of GrblPanel is (very very slowly) happening in the background.

tklus, 2017-01-27:
@MeJasonT, I am working on a pendant right now. I made up a quick instructable that describes how to use an F310 joystick as a pendant and I am also building up a new pendant using a CNC machined box and buttons and a usb joystick encoder like the ones used for usb arcade controllers. perhaps we should collaborate on this project. I will send you my email.

gerritv, 2017-01-29:
Fixed/implemented the last issues, I hope, on this one

closed 2017-01-29 by gerritv