gerritv/Grbl-Panel #85: Alarm state V1.0.9.10

Mac2712, 2017-01-03:
When you get an alarm should the status display not say alarm and turn red also reset button turn red?

gerritv, 2017-01-03:
What does the Status list show? Might have to click on Verbose.

Mac2712, 2017-01-03:
Morning Gerritv
The Status list shows
Alarm:1
[MSG:Reset to continue]
I also tried a homing timeout
this worked ok, got a Alarm:6 and the Status showed Alarm in red and the unlock turned red
Tried a Probing G38.2 timeout
this worked ok, got a Alarm:5 and the Status showed Alarm in red and the unlock turned red
Don't know any other Alarms to test let me know and I will test them. 
Hope this helps

gerritv, 2017-01-17:
I think I have them all now, thanks for reporting

closed 2017-01-17 by gerritv