gerritv/Grbl-Panel #86: Graphical Problems on 1920x1080 screen

1Metalguru, 2017-01-08:
With the newer versions of GRBL Panel, running on Windows 7 @1920x1080 resolution,  I have experienced a couple of screen related problems.  This has been a problem on at least the last 2 versions.

1.  The minimize button does not work.  There is no way to minimize the GRBL Panel window.  The only thing I can do to uncover the desktop is to move the window off the side or off the bottom of the screen.  The maximize button works normally

2.  On a 1920x1080 screen, the window is missing the bottom portion.  I can't see the lower part of the window, nor am I able to resize the window.  With the latest release, this appears to have gotten worse, I cannot see anything below the Rapid override buttons.

MG

[image]

gerritv, 2017-01-09:
The minimize button will re-appear in the next release.
I will attempt to reproduce your other situation, thank you for the clear details. My secondary monitor is 1920x1080 so fingers crossed.

The screen size was locked due to laser users maximizing the display and then complaining that their execution rate was slowing down. This was due to having to update more lines on the Gcode list. Tough to please everyone :-(

Mac2712, 2017-01-16:
I have 2 PC with 1920X1080 monitor and GRBL panel fits easily into the screen in fact 1 1/2 times. I would suggest you check you screen setup ie correct driver for both graphics card and monitor and you have the font size set to 100%.
Hope this helps

closed 2017-01-17 by gerritv