gerritv/Grbl-Panel #88: Manual moving always send G20 Codes

Miggi68, 2017-01-26:
Hello I am out of Germany and we are always working in mm.
So could you add an "initialisation" Code at the beginning "Set Units" [G21] for inch or [G20] for mm
and remove th unit code out of Manual control commands.
then at the beginning I could set the Units (as Default) to G20 and the System could work with mm (or if there is G21 in inches)

This would be very very very helpful.

Michael

Mac2712, 2017-01-26:
@Miggi68 
If you look on the right hand side you will see in the state box Units that will select Metric or Inch
Also to jog in metric you need to check the tickbox to the left of the directional arrows.
When you jog the code has G21 in the line sent to your controller 
Once these are set in GRBL panel they will remain even on

gerritv, 2017-01-26:
I think what might help is if I take the $ setting returned from Grbl and use it to set jogging defaults accordingly. Michael, Is that the intent you are asking? Or are you suggesting that I set things based on seeing G20 or G21 in the Gcode that Is sent to Grbl?

Mac2712, 2017-01-26:
Michael
Its G21 for metric and G20 for inches.

Miggi68, 2017-01-26:
I have to check this tomorrow.
But I am sure that I set the combobox on the right side to "Metric", but
When I use the arrow keys, the x,y,z values always increase 2.54 inches.

So, if I set the parameter on right side (or in GBRL) to metric, It would be
Wonderful, that by pressing an Arrow button, that the steppers move 1 mm ( or x 0.1, 0.01) and so on...
Here it is now 22:43 MECT ... I am looking for the answers tomorrow morning

Thank you all.

Michael

@Mac2711: you are right G20 is inch and G21 mm

Miggi68, 2017-01-26:
Ups ... what means closed

gerritv, 2017-01-26:
Ah, I think the issue is Grbl setting $13. This defaults to 1 for reporting in Inches. You need to set that to 0, then all reports from Grbl are in Metric.

pbtllc, 2017-01-26:
Why am i not able to bring up my GRBL setting? The "Get Grbl Settings" tab is not working.

jahnj0584, 2017-01-27:
send $$

gerritv, 2017-01-27:
@pbtllc some clues might help me?? What version of Grbl? Are you connected to Grbl? What is in the Status area?

pbtllc, 2017-01-27:
It is the latest updated  one.  Why does none of the older older ones not work too?

Chris Abbott
Co-Owner at
www.poorboytactical.com<http://<br/>www.poorboytactical.com>

gerritv, 2017-01-27:
I can't help you if you can't provide information.

Miggi68, 2017-01-27:
O.k. I tried now ..
If i set $13 to 0    (Report iches = false) then I got 2.54 mm / joggle press
If I set $13 to 1  (Report Inches = true) then I got 1.000 mm / joggle press

Is it possible, that this has the exact wrong order?

greetings Michael

Miggi68, 2017-01-27:
NOW all is working : 
=============
My "Problem" was to set the checkbox [x] Metric on Settings page at [Jogging] groupbox.
And now also $13 =0 works fine .. I didn´t saw this Checkbox , sorry

Summary:
=======
German Users should set 3 different Things:

[Mainpage]             Units = METRIC
[Settings  ]              [x] Metric   at Jogging area
[Grbl-Parameter]    $13 = 0

Michael

closed 2017-01-27 by Miggi68

gerritv, 2017-01-27:
@Miggi68 Glad it is working for you now. The combination of options can be confusing!
re: 

> P.S. I forgot to tell that I use Grbl 1.1 - the latest.
> But because of pbtllc´s Problem?!, another Software also didn´t work with the latest grbl.... perhaps you also have to update your grbl Panel.
GrblPanel works for 0.9x and 1.1, it doesn't work with 0.8 or earlier. It is also common for users to connect to the wrong COM Port. Gives that symptom but since they didn't actually answer the questions I can't tell.

pbtllc, 2017-01-27:
When I get home I'll try all these suggestions  and take some screen shots. Thanks for the help. I can't remember the latest version,  but I think is is 1.0.?.12 that came out several days ago. Thanks again.

Chris Abbott
Co-Owner at
www.poorboytactical.com<http://<br/>www.poorboytactical.com>

gerritv, 2017-01-27:
@pbtllc It is the Grbl version that is important. If it is not 0.9x or 1.x (e.g. 0.8) then it won't work. Also there are Grbl variants that don't work. A more likely scenario is that you are not connecting to Grbl at all but to another Com port on your PC. Click on the drop down showing Com Ports to see if there is more than one.

pbtllc, 2017-01-28:
vers. is 1.0.9.12
[image]
[image]
[image]

pbtllc, 2017-01-28:
[image]

gerritv, 2017-01-28:
@pbtllc Ok, will take a  look later today. Thx

wind-rider, 2017-01-28:
@gerritv: not meaning to hijack the thread, but with the latest grbl v0.9 and grbl-panel 0.9.11 and 0.9.12 I have the same problem that grbl-panel can't read the grbl settings into the settings table. When switching on verbose you can see the settings in the feedback from grbl though.

gerritv, 2017-01-29:
@pbtllc , @wind-rider I fixed the No Settings problem.