gerritv/Grbl-Panel #95: Win 10 IoT core

cacope, 2017-02-23:
I was wondering if this program would run or be converted over to win 10 IoT core on a raspberry pi and a 7 or 10 in screen. This would make a nice small cheap controller for a desktop cnc.

gerritv, 2017-02-23:
It would mean converting to a UWP, which I did look at for a while. The big issue at the time was no serial/FT232 support but that was eventually provided. In the end it is a lot of work to end up on a small display with fewer GUI features.
I don't have any motiviation for doing this myself, for about the cost of a Raspi and display you can get a used laptop.

closed 2017-02-23 by gerritv

tklus, 2017-02-23:
I have been using this little laptop from Walmart with great success. On sale right now for $108 with free shipping. 

I asked the same question a while back and another fellow suggested the same to me.

By the time you buy the rpi, monitor, conversion cable from hdmi to vga (most old cheap monitors don't have hdmi), keyboard, power supply for the pi. You are getting really close to the $100 mark.

The cambio runs windows 10 has wifi built in and 1 USB port. I got a USB hub so I can connect multiple devices and it works great.

I setup a drop box account on it and have a more powerful computer I use for programming. I post the program to drop box and t magically appears on the laptop :D

And as a bonus, I have a computer I can take out in the field for programming my rc race quadcopters. And it's touchscreen 

https://www.walmart.com/ip/RCA-Cambio-10.1-2-in-1-Tablet-32GB-Intel-Atom-Z3735F-Quad-Core-Processor-Windows-10/45804385

gerritv, 2017-02-23:
@tklus  Bingo, what a deal that is. And touch as well, which is not supported on Win IoT yet.