gerritv/Grbl-Panel #100: Jog controls wonky

1Metalguru, 2017-03-17:
Just doing some testing on a new machine, and I have a really weird problem with jogging on the Y axis.

When I jog Y + or -, with the jog distance set to 10mm, it actually moves a random amount, from 20 to 60mm.  The command sent to GRBL is: $j=G91G21Y(+/-)10.  If I send a manual G91G21Y10, it move 10mm as it should.  Only the Y jog controls are acting weird.

Both the X and Z axes behave normally.

Any ideas

I will sens a video shortly.

MG

1Metalguru, 2017-03-17:
Here's a video:

Ok, here's not a video.  Won't let me post it.

gerritv, 2017-03-17:
What does it show in the Work display? The correct values or the wrong one?

Is the Feed Rate weird for Y axis?

I tried it using 1.1e (2017 January). I don't have a physical machine though so only going by what Grbl sends back.

1Metalguru, 2017-03-17:
H Gerrit

The work display shows the actual movement, ie if it moves 30mm, the
display shows 30, even though it was only commanded to move 10mm.

Feed rate appears to be what it says for both jog and manual entry.  The
only thing I changed recently was the default jog speeds, went to
2000,1000,500,100.

I am using GRBL 1.1f, and GP 1.0.9.13, Windows 7.  Just rebooted and
restarted GP, same thing.

I am going to try reburning 1.1f and see if anything changes.

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

Regards,
Darrell Norquay - Calgary, Canada
3D Tech Works - 3DTechWorks.ca
403 701 2005 - ndarrell@telus.net

1Metalguru, 2017-03-17:
Hi Gerrit:

I just found the problem.  I had previously changed the X and Y repeat
rates to eliminate the stutter when holding the mouse button down.  I set
the X repeat rate to 3.5, but I accidentally set the Y repeat rate to 33.5
Changing it back to 3.5 solved the issue.

Not sure why this did this, it may still be a small bug somewhere.  If the
repeat rate is set too low, the machine "stutters" or "hiccups" when you
hold down the mouse on a jog button to move long distances.   It goes
"moooooove, stop, moooooove, stop, moooooove, stop"  Increasing the repeat
rate fixes this, but 33.5 is obviously way too high.

Thanks for your prompt response.

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

Regards,
Darrell Norquay - Calgary, Canada
3D Tech Works - 3DTechWorks.ca
403 701 2005 - ndarrell@telus.net

gerritv, 2017-03-17:
Glad you figured it out. Grbl works at its best when fed faster than what the machine can actually move. Anything less often results in that stutter effect.

closed 2017-03-17 by gerritv