gerritv/Grbl-Panel #101: Feature request - Macro auto-repeat / loop feature -> canned cycles

organetic, 2017-03-22:
I have just started to use GRBL panel, and it seems to me now that this is the real standard for GRBL sender, G-code interpreter.

I just miss a feature related to macros: The possibility to have a field, where you select how many times you want the macro to be repeated.

This would open up many new possibilities.

Also, is it impossible to implement more axis? This question is not related to GRBL itself, but with the GUI.

I guess it is just a matter of enabling more axis in the options, just like GRBL controller 3.6.1 has.

There's a number of GRBL firmware options on the web. Communication protocols and commands are pretty much the same. Only the ammount of axis changes.

Keep up the good work. Your job is paramount to the community ;)

Compliments from Portugal :)

gerritv, 2017-03-22:
Thank you :-)
Where can I get a multi axis copy of Grbl Controller 3.6.1? It has not been supported since 2014 so not sure what it will work with.

LETARTARE, 2017-03-22:
Hello,
You can try an very older version [here](https://github.com/LETARTARE/GrblHoming/tree/Letartare/GCV-3.6.1-T4)
Good night.

organetic, 2017-03-23:
Hi! Indeed, that's exactly the one I used, from Letartare's repository.

Letartare was one of the pioneers on GRBL ports for the 2560, with 4 axis enabled.

I have used them once, without any problem notice. I never had to move the 4 axis at the same time anyway.

As a picture is worth a thousand words, here you have GRBL controller 4th axis options :)

Any word regarding the possibily to define the number of times we would like a macro program to be repeated?

[image]

gerritv, 2017-03-23:
@LETARTARE and @organetic  thank you for the links. I downloaded and had a look.
Not too difficult, it is a matter of deciding to wait for the Arm version of Grbl and my rewrite in C# or code it up but not test it (I am not interested in buying a 2560).

Re: the macros, I will have to finish a related effort first, namely putting the macro gcode into its own queue so that it waits for Ok from Grbl. Presently it doesn't, just hoses it all out at once. Usually works ok but won't if the buffer fills up and you have to actually wait to send the next line. After that it is easy to just load x copies into the buffer.
I am working on a machining project at present, should be able to do some coding in 1.5 weeks maybe sooner.

organetic, 2017-03-23:
Thx for the update Gerrit. 

I'm actually using GRBL 0.9j with and Arduino UNO and 4 axis.

It works really fine.

This is the link for the UNO version:
[https://github.com/mlambm/grbl](https://github.com/mlambm/grbl)