gerritv/Grbl-Panel #102: Feature Request - Lathe configuration

MeJasonT, 2017-03-23:
Hi Gerrit,

I have a seed burning a hole in my tiny mind.  I keep looking at my lathe and pondering as to weather it would be worth automating it as well.  Grbl-Panel is designed around the axis of a milling table, plasma table, laser cutter etc.  as far as i can see the Y axis would become obsolete for 2 axis turning but could be used for indexing or cutting designs into a tube.  
If i were to use a 4th axis on my plasma table and swap the Y axis wiring and control to the A Axis motor then i can see amazing possibilities for producing cylindrical designs and cutting notches for pipe welding etc.
I think grbl-panel would/will require new settings (for a different machine) perhaps it wont care as the settings are based on the grbl controller and not the GUI.

perhaps the GUI could have two or more machines associated with it and upload settings to the grbl board on startup or on selection.

I would be interested in your feedback.

Ps. Im already using two laptops for my Miller and plasma table + a defunct mac for my 3d printer.

gerritv, 2017-03-23:
I have similar aspirations and am doodling up a rewrite in C3 that is less messy. Present code is a result of good design start subsequently messed up by blending GUI and implementation details in same module :-(

I will make a start on coding soon as the ARM version of Grbl is coming soon. I was thinking of coding using Xamarin to add Android at least but will instead just focus on separation of layers for now.
The above should result in a more flexible GUI support as the layout changes for lathe vs mill etc are mostly cosmetic.

 I only have so many hours in the day and learning a whole new technology whilst also progressing with my hobbies is too much.

Are you on Win7 or 8 or 10?

MeJasonT, 2017-03-25:
I'm still using the only reliable operating software Microsoft ever wrote ish.
good old Win 7

MeJasonT, 2017-03-25:
My 18" desktop screen is not a touch screen so the tiles in windows 8 designed for use on mobile devices is a waste of time and i have heard of many users not satisfied with 10 either.  i'm sure the amount of updates that have been sent out over the last month MS must have rebuilt my OS by stealth by now.

gerritv, 2017-03-25:
I really prefer Win 10 now but 7 was great. Of course I enjoyed 3.1 as well :-)

The reason for asking is because there was another discussion about using Xamarin and XAML to re-use the code for Android as well. I think I wil focus on WPF/XAML/C# so I can cover Windows 7 as well.

MeJasonT, 2017-03-26:
On behalf of us dinosaurs, thanks.

richardanmmol, 2017-05-17:
Do you have source for Android?

tklus, 2017-05-24:
I am also interested in using GRBL for lathe, I just picked up a manual lathe and considering converting it :)

Thanks your hard work :)