gerritv/Grbl-Panel #107: Future Request for file name

wlaw, 2017-04-22:
OK, so i'm working with my machine this morning and have 2 files to cut, left side and right side.
I just finished cutting one of them and was ready to cut the other. Is there a way for me to tell the file that just finished? :-) 

I have been looking for it and cant seem to see it. 
My wife says that's what I always say :-)

So, is it possible to add the filename that is loaded on the screen? Maybe next to Message.
[image]

Again, thanks for making Grbl-Panel so great.  

Kind Regards

Bill

wlaw, 2017-04-24:
OK, so for now I added this to my VCARVE grbl post processor.
";[TP_FILENAME]"
This adds the file name at the top of the G CODE file.
[image]

gerritv, 2017-04-25:
I'm puzzled. Above the file control icons is the area where file name is supposed to be. Was it there when you loaded the file with the Folder icon? Since the Run/Start button is not highlighted I am guessing that you don't have a file loaded?

Also noticed that you must be using quite an old version, the new version has a Single Step icon next to Start/Run

wlaw, 2017-04-25:
[image]

1.0.9.7 - A bit behind, but I get tired of loading the macro's I use all of the time.

wlaw, 2017-04-25:
I guess 9.7 is a bit old, I will upgrade tonight and let you know.
I just hate to break something that's working :-)

Thanks for your help.

gerritv, 2017-04-25:
Copy the user.config file from %localappdata%\GrblPanel 1.0.9.7 subdirectory to the new one created after you run the new copy first time.

wlaw, 2017-04-25:
:-)

wlaw, 2017-04-26:
I see it now!

Thanks @gerritv 

[image]

closed 2017-04-26 by wlaw