gerritv/Grbl-Panel #108: GRBL Panel crashes when soft limit is enabled

alfredabounasr, 2017-04-22:
when soft limit is enabled and when the value of any axis reaches the soft limit value, GRBL Panel crashes

gerritv, 2017-04-25:
You will need to provide a few more hints, e.g. 
what version of Grbl, 
what version of GrblPanel, 
what are your limits set to, 
do you have Hard limits enabled,

Can you screen shot the Status area after enabling Verbose, so I can see what Grbl is sending back.

alfredabounasr, 2017-04-27:
Hi,
I'm using the latest grbl on arduino with the latest grbl-panel software

alfredabounasr, 2017-04-27:
I'm using the latest grbl on arduino and the latest grbl-panel.
Hard limit is enabled
The limits are set to 200mm for testing purposes and all are positive
numbers.

gerritv, 2017-04-27:
Is it 1.1f or 0.9 Grbl?

alfredabounasr, 2017-04-27:
It's grbl 1.1f

gerritv, 2017-04-27:
I am not going to play 20 questions any longer. Read my previous requests, no info, no fix as I can't reproduce it with the set up that I have.

alfredabounasr, 2017-04-28:
Hello,

here are the hints that you asked for:

GRBL version : 1.1f.20170131
Grbl-Panel Version : 1.0.9.13
Limits set to 200mm
Hard Limits is enabled

Thank you and sorry for the inconvenient.

regards,
Alfred.

Alfred Abou Nasr / Founder and General Manager
alfred@frictionlb.com / +961 3 113723 <009613113723>

FRICTION
frictionlb.com

[image: Twitter]  <https://twitter.com/frictionlb>   [image: Facebook]
<https://www.facebook.com/frictionlb>   [image: LinkedIn]
<https://www.linkedin.com/in/frictionlb>   [image: Instagram]
<http://www.instagram.com/frictionlb>

gerritv, 2017-04-28:
Yes, but I also asked for a screenshot of the Status section with Verbose turned on. Since I can't duplicate the problem I need to see what Grbl is sending that makes GrblPanel crash.

alfredabounasr, 2017-04-28:
The screenshot is attached within the last email i sent. Grbl is returning
 "error:15"

gerritv, 2017-04-28:
Unfortunately images don't come through when responding by email :-( Knowing it is error:15 now I know it is Jog command that it fails on. 

I know what is wrong with the code, will fix it this weekend and push new version.

alfredabounasr, 2017-04-28:
Thank you very much and I appreciate your hard work.

gerritv, 2017-05-01:
I added 3 missing error messages. Fixed in 1.0.9.14

closed 2017-05-01 by gerritv