gerritv/Grbl-Panel #110: crash on usb unplug

ScottTu, 2017-05-13:
unhandled error if you unplug the usb cord after being connected.

gerritv, 2017-05-13:
I can't duplicate this.

ScottTu, 2017-05-13:
next time it happens I'll get some more error info. I'm using a desktop cnc with an arduino controller.

electrokean, 2017-05-13:
It is very likely dependant on the specific USB serial interface (FTDI, 16U2, CH340) and version of Windows driver you have.

ScottTu, 2017-05-13:
could be, its not a critical bug but I've had it do it while resetting the machine or powering it off after cutting.

ScottTu, 2017-05-13:
[error.txt](https://github.com/gerritv/Grbl-Panel/files/998969/error.txt)
win7 64, ch340 serial adapter on the arduino.

gerritv, 2017-05-14:
I'll add an exception handler for Denied, that should take care of the issue.

ScottTu, 2017-05-14:
thanks and thanks for the great program too.

gerritv, 2017-06-23:
Fixed

closed 2017-06-23 by gerritv