gerritv/Grbl-Panel #111: Feature Request: Set axis to a specific value.

Jimboe98, 2017-05-15:
Currently the only option is to set an axis to zero or set all axes to zero. Often it's necessary to touch down on the component at a non zero height  and set zero relative to this height. For example suppose the stock is 5mm thick and the origin is at the bottom of the stock. You touch the tool on the top of the stock and then need to set Z to 5mm. But you can only set 0. So the way round is to set it to 0 then drive the machine to a free area and drive down to Z -5mm then set 0. This is prone to error. It would be better allround if you could set each axis to any value not just zero.

Jimboe98, 2017-05-15:
Sorry, It's already in the list below.

gerritv, 2017-05-15:
Let's leave this one open, it is a reminder for me to get on with adding the feature :-)

gerritv, 2017-05-31:
I am looking at finally implementing this. The value typed in would set the currently active work offset to the value. Is that your understanding?

Previous discussion is here: https://github.com/gerritv/Grbl-Panel/issues/12

tklus, 2017-06-07:
This would also be a nice feature for setting "x" diameter when setting lathe tools.

Would it be difficult to set up a GUI that is for lathe control?

tklus, 2017-09-27:
Gerrit, are you still working on this feature? I have a buddy who just set up his milling machining and he is using your gui :) we love it!

is there a way to set up tool offsets too? this would be very helpful for jobs that require multiple tools.

gerritv, 2017-09-27:
I started to but ran into some challenges with stopping the Grbl responses to '?' commands from overwriting the value being entered. I will take another stab at it. The whole thing is pointing me to doing a rewrite due to the code getting messier as time goes by. The separation between gui and objects is now seriously blurred making some things a challenge.
(had to build 3 decks and redo a garden this summer so not much hobby time :-)

tklus, 2017-09-27:
Thanks for the reply! I hear ya, summer is a busy time... I can't wait for the rain to hit so I can get back to hobby stuff. Let me know if you need any help testing.
Thanks
Tim

gerritv, 2017-11-09:
Implemented.

aeleus, 2017-11-10:
When I try to type a number in the display it gets overwritten as quickly as the refresh interval. If I change the refresh interval to give me time to type something in, the results are unpredictable. I type "5", hit enter, the command history show G10 L2 P1 5, but the display shows -0.1969?!

gerritv, 2017-11-10:
It should be showing as 'G10 L2 P1 X5 if doing it on X axis.
Two things going on: 
1. there is some interaction between Grbl's param $13 Report inches and the Metric/Inch G20/G21 behaviour. If you flip the State/Units drop down and back it should correct itself. I will try to figure out what is wrong there
2. I can't reproduce the behaviour you described re: overwriting your typed info. What is your Status Report setting ($10) set to, mine is 3

aeleus, 2017-11-10:
I left out the X. The command history shows G10 L2 P1 X5.

Report setting ($10) is set to 3.

State/Units was set to Metric. Changed it to Inch. Now, if I type a number and hit enter, it changes to the negative after the poll interval hits.

E.g.:
Change poll interval to 5000 (otherwise I don't have enough time type anything)
Type '3' in the X position box and hit enter
Position displays just '3' for the X position
Command history immediately shows 'G10 L2 P1 X3'
After the poll interval hits, the display changes to '-3.0000'

Laptop running Windows 10 (1709)

gerritv, 2017-11-10:
I just tested again
Windows 10 (1709)
Poll interval at default
When I click in one of the position boxes and do nothing else, they box stays empty. (i.e. the focus has to stay on that entry box)
So I can't reproduce your symptoms, makes it hard to fix :-(
I do have to change one thing, allow entry of '-' so that you can go above or below the present offset position.

gerritv, 2017-11-13:
Fixed to allow entry of '-'

closed 2017-11-13 by gerritv