gerritv/Grbl-Panel #112: Feature Request: Add the ability to run Mach3 Style Pendant

sonnylowe, 2017-05-15:
I'd love to be able to run a [Mach3 Style Pendant](http://www.ebay.com/itm/iMach3-P1A-USB-CNC-MPG-Pendant-for-Mach3-LinuxCNC/162491243217?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2060353.m1438.l2649) or [Universal Pendant](http://www.ebay.com/itm/Universal-CNC-4-Axis-MPG-Pendant-Handwheel-Emergency-Stop-for-Siemens-US-STOCK/252349701581?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2060353.m1438.l2649) with this software. Although I make all my setup adjustments on my mini mill manually, this would be great for something like the Openbuilds XL C-Beam Machine since you can end up pretty far away from your laptop or tablet of choice will setting up your stock. 

If this is a possibility, I'd be willing to provide a pendant to help with the integration.

Thanks,
Sonny

gerritv, 2017-05-17:
I have asked VistaCNC for info on their interface. The challenge with the Universal one is that it needs a controller with PC board etc. That would be beyond my available time although its price point is in my range :-)

sonnylowe, 2017-05-17:
This sounds very promising, thank you for looking into it 👍 

Haha, I agree, the cost of the universal unit is very attractive...it's why I included it ;-) But i certainly understand how it's complexity can make it unviable... 

If you get the information you need for the Vista Pendant let me know and I will be happy provide you with the pendant.  

Thanks,
Sonny

wlaw, 2017-05-17:
That would be a dream come true. 😀
I would be happy to get a pendant and test also.
Let me know please.

gerritv, 2017-05-20:
No response from VistaCNC support, trying sales.

gerritv, 2017-05-21:
Info is on the way. Looks like mostly a keymapping exercise.

sonnylowe, 2017-05-21:
Great news, let me know if you'd like me to either provide the pendant or test it for you, I'm good doing either!!!

Thanks agin, 
Sonny

sonnylowe, 2017-05-24:
I just ran across this [LITTLE BEAUTY](https://www.cncmasters.com/product/mx-keypad/) today, is configuring a standard Keypad an option for pendant control!?!?

tklus, 2017-05-24:
So... I have been working on making my own pendant but have not had time to finish it up. it uses an arcade usb interface and some buttons. I have it roughly planned out but need to execute the design. I was originally going to laser cut the panel face out of wood and make a wooden case for the card and switches but have decided it might be nice to just 3D print one. I need to finish up the design but would be happy to share when I am done.

then you can use Antimicro to map the buttons to keyboard commands.
https://github.com/AntiMicro/antimicro

here is the arcade interface.
https://www.amazon.com/Easyget-Arcade-Encoder-Joystick-Fighting/dp/B00XHRIKLE/ref=sr_1_fkmr0_3?s=toys-and-games&ie=UTF8&qid=1495638252&sr=1-3-fkmr0&keywords=arcade+usb+card

[image]

tklus, 2017-05-24:
see this post as well :)
https://github.com/gerritv/Grbl-Panel/issues/84

tklus, 2017-05-24:
this might be an easier solution as you suggested, and we should be able to map the keys with antimicro :)
https://www.amazon.com/dp/B01N5CM4N7/ref=sxr_pa_click_within_right_4?pf_rd_m=ATVPDKIKX0DER&pf_rd_p=3008539542&pf_rd_r=13PNX0ZRSVQP615PVCJX&pd_rd_wg=4FYIo&pf_rd_s=desktop-rhs-carousels&pf_rd_t=301&pd_rd_w=S7Nyc&pf_rd_i=20+key+keypad&pd_rd_r=DEPXJ7F23W2MGPDRGGR1&psc=1

tklus, 2017-05-24:
even more interesting :P 
https://www.amazon.com/dp/B01N5MF1X2?psc=1

tklus, 2017-05-24:
I ordered one of the corded ones and will test it out this weekend. good idea!

gerritv, 2017-05-24:
The present mappings are: https://github.com/gerritv/Grbl-Panel/wiki/Key-Mappings If I need to add more let me know.

wlaw, 2017-05-24:
I have been using a USB game controller as a pendant for a while now. One
of my first contacts with @gerittv was about key mapping. The program I use
for key mapping is keytojoy. It works well and I have very little trouble
with it. So a usb keypad is not for me. But the pendant with the dial is
just what the Dr. Ordered. The ability to change speed by changing speed
you turn on the dial is fantastic.

If your interested in more about the game controller I would be happy to
help.

In any case thanks for all your work.

Grbl-Panel/wiki/Key-Mappings If I need to add more let me know.

sonnylowe, 2017-05-27:
Okay, I got my USB Numeric Keypad yesterday and it was completely "Plug & Play". Most of the Key-Mapping features worked. Here's a pic of my quick and easy labeling of the buttons that currently work (with the exception of the "0" button), I thought the "ins" button would function as "insert"!?!?

I might be me doing something wrong as fas as the other mapping not working, so I'll play with it a bit and get back you guys with my thoughts. 

Overall though, this is a great addition!!! 

[image]

wlaw, 2017-05-27:
Awesome,

I know I would hate not having something.
How about the repeat rate?  is it one move per push or will it repeat.

On the game controller and joytokey I can control this function.
for instance, I have set X and Y to repeat if held down, but Z will not
repeat.
I must press Z each time I want movement.

On a side note, are you running screws on a shapeoko?

I have the 1000mm screws, and have wondered if i want to mess with it.

If you are, and its not to much trouble, can you send me pics?
And let me know what you think of it?

sonnylowe, 2017-05-27:
wlaw,

> How about the repeat rate?  is it one move per push or will it repeat.

One push one move, however, hold the button down and off is goes ;-) As long as the machine can keep up all is well, but let the request get ahead of the machine and it keeps running after the button is released. 

> On a side note, are you running screws on a shapeoko?

I am actually playing with on My [OB Minimill ](http://openbuilds.org/builds/openbuilds-minimill.5087/) the OB Parts Store just released it today. Mine was an ongoing collaboration with Mark during the design stage. It does use lead screws with the longest being around 350mm. I original intent when requesting the pendant however was to run it on something the OX or C-Beam XL where tablet/PC access is a bit more limited. This is my current setup...

[image]

> I know I would hate not having something.

I cant stress enough how happy I am with the ease and usefulness of the current setup, so anything beyond that is shear bonus in MHO. What said I'd like to figure out how to get the following working utilizing some of the remaining (unused) buttons on the keypad. 

 It would be great to have the following functions on the keypad, "Home Cycle" Zero XYZ, Start/Resume, etc.

 I think it would also be cool if at the point where you're done setting the machine you could use the "number lock" feature to open up any battery of functions, all driven towards running the machine vs setting it up. Like changing feedrate and spindle speed on the fly, also have all of the Ctrl+ function currently listed on the Key Mapping page available while in Number Lock mode!?!? 

Just some food for thought, but again, let me please reiterate how happy I am with the current functionality and if that never changed it wouldn't bother me a bit.

Not to beat a dead horse, but I still want that pendant and the hand wheel...that would be perfection :-)

tklus, 2017-05-30:
Hey guys, so I tested my key pad and now I see the limitations as compared to the "joystick" and joy to key.
You are basically stuck with jog functions and toggle for speed and distance. You cannot map keys with the keypad.

I will continue designing my push button pendant to be used with the USB joystick encoder.

Thanks
Tim

I find it quite useful but would be nice to have zero

gerritv, 2017-08-01:
Sadly VistaCNC never sent me the info. Putting this on hold unless someone needs more items mapped to key strokes.

There are some really good solutions offered in this Issue chat, thank you.

closed 2017-08-01 by gerritv