gerritv/Grbl-Panel #114: Unhandled exception while connecting via USB

RSG88, 2017-06-04:
Initial connection to a new controller,  I am also new to using Grbl-Panel, any advise?
[image]
[image]
[image]

gerritv, 2017-06-04:
You are using an old version of Grbl-Panel with what I am guessing is Grbl 1.1.

RSG88, 2017-06-04:
It is working now on latest revision, Thank You so much for your quick response and help!

closed 2017-06-04 by gerritv