gerritv/Grbl-Panel #115: Manually control stepper motor with joystick

Camerutttt, 2017-06-13:
Hey I was wondering if it is possible to manually control a stepper motor with a joystick.

Right now, all I am able to do is to make the stepper motor turn for a certain number of revolutions when my joystick is pushed in a certain direction.

I want to make it such that the stepper motor will keep on turning for as long as I push my joystick and stop when I release my joystick.

tklus, 2017-06-13:
I am currently using a joystick to run my machine and I can set the step distance to .1 inches and just hold the joystick and it will feed until I release the joystick.

there is a setting for the repeat rate when the joystick is held so you might be able to adjust that.

are you using the latest version of GRBL and GRBL Panel?

Camerutttt, 2017-06-13:
I think I got the latest version since I just downloaded it today. 

Pardon me, but I am new to GRBL. So from what you have done, I assume you map your joystick to the computer arrow keys? Since that is how I do it. My joystick is the 1 with 4 switches, so when you push it in a particular direction, 1 of the switches will be activated. Also, all I have to do is change the distance to 0.1 inches and set it on repeat rate?

tklus, 2017-06-13:
You are correct, I am just using a joystick and mapping to the key strokes. I will have to look at my repeat rates but that is something you can try.

Camerutttt, 2017-06-13:
So all I have to do in the GRBL Panel is to adjust the distance to 0.1 like so: http://imgur.com/WMzKalX?

and to play around with the repeat rates and check the Enable keyboard arrows like so: http://imgur.com/9jYfjsr?

tklus, 2017-06-13:
Yup, give it a shot. It works with my machine....

Camerutttt, 2017-06-13:
Alright thanks for your help. Unfortunately, all my stuff arent with me right now so I will let you know how it goes later on.

sonnylowe, 2017-06-13:
Wow, this is good stuff, thanks for sharing!!!

tklus, 2017-06-13:
what key mapper are you using? that could be the issue as well. maybe there is a setting for repeat in that too.

I am using antimicro for my key mapping

Camerutttt, 2017-06-14:
Hey I just tested it and it appears to be running quite smoothly. However, when I release the joystick, the stepper motor will still continue to run for a few seconds before stopping. So for this issue I should play around with the repeat rates?

I am using JoyToKey and yes, there is also a setting for repeat too.

Camerutttt, 2017-06-14:
I have also attached another stepper motor for the other axis. So now I can control 1 stepper motor by moving the joystick in the x-axis and the 2nd stepper motor by moving the joystick in the y-axis. However, when I move the joystick diagonally, both stepper motors movement are very jerky.

Camerutttt, 2017-06-14:
Also, is there a way to have a return to zero position button using an external switch?

tklus, 2017-06-14:
You can set up macro buttons in GRBL Panel and maybe we can request to have these macro buttons mapped to certain keys on the keyboard.

Camerutttt, 2017-06-14:
I know that I can set macro buttons but I am unsure of how to map them to my keyboard

tklus, 2017-06-14:
This would be a feature we would need to request from Gerrit...

Camerutttt, 2017-06-14:
So as of now there is no other way to do this? Maybe through something like UGS?

gerritv, 2017-06-14:
Adding some more key codes is easy, can someone pick values? I have other features to add but those are taking a bit more thinking.

tklus, 2017-06-14:
How about 1-9 number keys? or however many macro buttons are available, not sure off the top of my head how many there are.

Thanks for the support!
Tim

gerritv, 2017-06-23:
Implemented, macros map to F2-F6

closed 2017-06-23 by gerritv