gerritv/Grbl-Panel #116: Feature Request: configurable rounding of decimal places in UI

aeleus, 2017-07-05:
It would be great to have the option to set how many decimal places the UI displays.

My CNC setup is only capable of resolving to thousands when working in inches. E.g., inputting X1.000 may displays as .9997. I spend a lot of mental effort rounding in my head. The extra decimal place doesn't provide me useful information - just more work.

Thanks!

gerritv, 2017-07-05:
Would you want to truncate or round up/down? Some of the 'error' is due to Grbl operating internally in metric and converting to 'inferial' if you set it to do that.

aeleus, 2017-07-06:
Rounding (up or down as appropriate) would be more helpful. That way .9997 displays as 1.000 and .9993 displays as .999

ljbuller, 2017-07-26:
The rounding would need to be user selectable because .9997 is not always appropriate to round to 1.000
like gerritv said, using a metric lead screw (or Grbl internal metric) and imperial coordinates will give you +/- .0001.  I have changed my coordinate to display 4 decimal places.

aeleus, 2017-07-27:
I suggest a check box to enable rounding the displayed number that, once enabled, allows the user to select the number of decimal places and whether to round or truncate.

gerritv, 2017-08-01:
I will be implementing rounding to a specified number of digits. Not really comfortable truncating data because you really can lose meaning in the process. E.g. rounding .9997 to 1.000 is a safer alternative to allowing someone to truncate to .9. Yes,  I could do a check to see if the number of digits is 'large' enough but each bit of fiddly logic introduces other side effects.

Sunnysky, 2017-08-27:
Is it possible to add a feature for dual Y motor XY Gantry with dual drivers to have a 4th axis?   Grbl Controller 3.6.1 has this where you can assign to the 4th axis to any 3rd axis then invert any of the 4 axes.
I have grbl 1.3f with Genuine UNO and a 4 port Motor shield.

gerritv, 2017-11-09:
Implemented

aeleus, 2017-11-10:
Thank you for implementing! But, I cannot get it to work. :-(

I change the "Round to digits" from the default of 0 to 1, 2, 3, -1, -2, etc., but there is no change even when I click "Refresh" or restart grbl-panel.

What's the trick?

gerritv, 2017-11-10:
Not sure, it is working for me. I enter a number in that settings box, press Return and it changes on the Position tab.

aeleus, 2017-11-10:
Laptop running Windows 10 (1709)

Is there a way to dump the grbl settings?

gerritv, 2017-11-10:
Not directly. I use Snipping tool to capture a screen area, included in Windows 10.

aeleus, 2017-11-17:
I tried running on 1.0.9.16 on a different PC (Windows 10, 1709) with the same behavior.

Just tried 1.0.9.17 with same results.

:-(

[image]
[image]

gerritv, 2017-11-17:
Now I see why, you are running Grbl 0.9? I only implemented the enhancement in the 1.1 section of the code.

aeleus, 2017-11-19:
Thanks, gerritv. I realized that shortly after I posted my settings. I thought I had already updated GRBL.

All is good! Thanks for all the hard work!

closed 2017-11-19 by aeleus

gerritv, 2017-11-19:
I was nervous for a bit there, glad it is working for you. I might drop pre-1.1 support soon, I don't test with that version anymore.