gerritv/Grbl-Panel #117: Will only run single block

tbucket32, 2017-07-10:
When I load a .nc file, it shows up in the window.  I hit RUN and nothing happens.  If I keep clicking single block it will execute.

gerritv, 2017-07-10:
I can't reproduce that, when I load a file and press Start/Resume Sending the file starts up. Perhaps you have an alarm or error condition? That will show in the Status window and/or the Sts column of the GCode window.

tbucket32, 2017-07-10:
When I press start it just sits there and does nothing. I see no alarms or errors.  I can let it sit 30 seconds and it does nothing. Its like it trys to send and is waiting for a response but gets nothing so it hangs. But single block is no issue.  Maybe I'll reinstall and try again

Casey Linn

gerritv, 2017-07-10:
Is there anything in the Sts column after clicking Start/Resume? It should at least say Sent. A screenshot of GCode and Status sections would help me maybe see the reason.

tbucket32, 2017-07-10:
It never says sent...status column is always blank. Ill try to get screen shots tonight or early tomorrow morning.

Casey Linn

tbucket32, 2017-07-10:
Attached is a picture of the screen after i have pushed start.

Casey Linn

tbucket32, 2017-07-10:
I just deleted the files and downloaded them again. Its doing the same thing.

Casey Linn

gerritv, 2017-07-10:
Unfortunately the screen shot didn't make it to Github :-(

tbucket32, 2017-07-11:
Hopefully the picture works this time.
[image]

gerritv, 2017-07-11:
it almost looks as if  the file has not finished loading, the File and Start icons should be darker. If light coloured then they are not enabled. I will look over the code later today

tbucket32, 2017-07-11:
The code is only a couple lines. It draws a simple 5mm square box.

In the image, I had clicked the load button and selected the file then hit start....that is when the buttons when light coloured.

Casey Linn

tbucket32, 2017-07-11:
Could it be caused by only having gbrl 0.8?  Its a Laseraxe controller that has pins D11 & D12 hard wired for the old gbrl.

Casey Linn

gerritv, 2017-07-12:
Ah, yes I do not support Grbl 0.8 although usually the symptoms are more obvious. I should perhaps put something in the code to detect and warn users.

You should consider recompiling Grbl 1.1 with the pin settings reverted back to 0.8 layout. Changing #define VARIABLE_SPINDLE to a comment on line 339 of https://github.com/gnea/grbl/blob/master/grbl/config.h will swap the pins back to 0.8 layout.

tbucket32, 2017-07-12:
If i do that, will i lose pwm control of the laser?

Casey Linn

gerritv, 2017-08-01:
I am not sure what you would lose, if anything. PWM was added in 0.9 but the pin map had to change. So your board must already deal with that? 
Plan B would be to alter the cable(s) connecting to the board so it woks with 1.1 versions.

closed 2017-08-01 by gerritv